supermerill/SuperSlicer #4031: Crash when opening the filament cooling tab

sprior, 2023-12-31:
### What happened?

With version 2.5.59.5 if I open the filament settings I can get to the basic Filament tab, but if I click on the cooling tab the application crashes with an internal error: invalid string position.

### Project file & How to reproduce

open the filament settings I can get to the basic Filament tab, but if I click on the cooling tab the application crashes with an internal error: invalid string position.

### Version

2.5.59.5

### Operating system

Windows 11

### Printer model

voron 2.4 with Stealthburner

sprior, 2023-12-31:
It looks like the profile it was crashing on had a parameter named min_fan_speed which now that I have saved the profile has disappeared and I am no longer seeing the crash.  Is this a deprecated parameter and that's not being handled well?

linux-paul, 2023-12-31:
I removed the parameter from the active profile, but it didn't help. Then I created a configuration bundle, set up on an empty environment and reinstalled the bundle. This has worked so far. Many thanks for the tip

supermerill, 2023-12-31:
can i have a way, like a project or a config to import to reproduce the crash?
Or a zip of a configuration directory...

linux-paul, 2023-12-31:
[SuperSlicer__.zip](https://github.com/supermerill/SuperSlicer/files/13801177/SuperSlicer__.zip)
copy from my old data.

SinisterRj, 2024-01-21:
I'm still having problems with any of my profiles for my Voron 2.4 on version 2.5.59.6. Windows 11 64 bits here. 
[Profiles_SuSie_Backup_01_24.zip](https://github.com/supermerill/SuperSlicer/files/13999622/Profiles_SuSie_Backup_01_24.zip)
[image]
[image]

supermerill, 2024-01-23:
can you try the nightly?
To check if it's fixed, as i can't reproduce with the nightly.

SinisterRj, 2024-01-23:
Yes, I just need the instructions on how to do that.

SinisterRj, 2024-01-23:
Tested and it works! Version: C/C++ Nightly windows x64 #669: Commit [8b5bbe0](https://github.com/supermerill/SuperSlicer/commit/8b5bbe0434be26e0c49a5d1455927805c000d92a) pushed by [supermerill](https://github.com/supermerill)

sprior, 2024-01-23:
I'm running the Win64 nightly released when you fixed the double start gcode issue.  I can't test this particular issue anymore though because I refreshed the filament profile that was crashing.

closed 2024-02-14 by supermerill