supermerill/SuperSlicer #4032: A tool path outside the print area was detected on latest 2.

metzenw, 2023-12-31:
### What happened?

What is the problem?
The tool tip exits the print area after just a few layers of printing.
[image]

What did you expect?

I expected the tooltip not to exit the print area with such a small object.

### Project file & How to reproduce

To reproduce, use the v6_Cartridge_Straight filter STL and generate gcode for a Voron 2.4 R2 350mm bed.

https://github.com/nevermore3d/Nevermore_Micro/tree/master/V6/STL
[v6_Cartridge_Straight.zip](https://github.com/supermerill/SuperSlicer/files/13801828/v6_Cartridge_Straight.zip)

### Version

2.5.59.5

### Operating system

Windows 11 Pro

### Printer model

Voron 2.4 R2 afterburner

supermerill, 2023-12-31:
can you `save_as` -> zip the saved .3mf -> paste it here?
It's easier to share stl&settigns in one go

metzenw, 2023-12-31:
I just added a header to it in the description. The two links were squashed together, so it was unclear that it was there. Cheers!

supermerill, 2023-12-31:
It's an error from the arachne algorithm.
I'll add a little hack to remove these points for now.

closed 2024-02-14 by supermerill