supermerill/SuperSlicer #4034: Debug Alert while saving to SD card

vladi552, 2024-01-01:
### What happened?

See attachment. This happens sometimes while I save gcode file to the SD card. Several last version have this issue.
[image]

### Project file & How to reproduce

This is hard to reproduce.

### Version

2.5.59.5

### Operating system

windows 11-64

### Printer model

TEVO Black Widow

Alohacajun, 2024-01-02:
Can you provide the project file and associated configs that throw the error?

supermerill, 2024-01-04:
preference -> gui -> hide tooltips on slice buttons

It's a gui error with theses tooltips.
I can't understand the problem.