supermerill/SuperSlicer #4035: Fan speed does not respect default fan speed value if the value is lower then 35%.

MichalDuban, 2024-01-02:
**Bug description**
Fan speed does not respect default fan speed value if the value is lower then 35%.

**Expected behavior**
Fan speed should respect any value provided in the default fan speed.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Filament Settings'
2. Click on 'Cooling'
3. Set 'Default fan speed' to value lower then 35%
4. Set 'Max fan speed' to value lower then 35%
5. Go to 'Platter'
6. Import some model and slice it
7. Show 'Fan speed' 
8. Displayed fan speed is 35% even if the value was set to lower value (on higher values it is working properly)

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Linux
 - Version 2.5.59.6
 - Voron 2.4

**Additional context**
It is working as expected in SS 2.5.59.3

Nbeknel, 2024-01-03:
The variable for minimum fan speed has been moved to printer settings.
[image]

MichalDuban, 2024-01-04:
Thanks, I missed this note. It is working as expected.

closed 2024-01-04 by MichalDuban