supermerill/SuperSlicer #4036: irregular gcode display

CuredPrusa, 2024-01-02:
### What happened?

In any version of 2.5.59.x (which I currently use) the overview of gcode instances does not correspond to the actual position of the nozzle in animation. At the beginning of the layer, I don't see message "layer change" anywhere. The same goes for outer/inner perimeter and so on. There is some delay involved, or it's too advanced, I didn't investigate. I wonder if that can be fixed, it's a helpful troubleshooting tool. 

### Project file & How to reproduce

All my projects have this problem, but I didn't test with the most simple and low layer count projects. 

### Version

2.5.59.0-2.5.59.6

### Operating system

W10

### Printer model

Marlin2

senseisimple, 2024-01-06:
What I've noticed is that the vertical layer preview slider isn't corresponding with the line count slider. in some cases for a whole section. this is more pronounced when the "GUI > Controls > Sequential slider applied only to top layer" box is unchecked.

supermerill, 2024-01-09:
> What I've noticed is that the vertical layer preview slider isn't corresponding with the line count slider. in some cases for a whole section. this is more pronounced when the "GUI > Controls > Sequential slider applied only to top layer" box is unchecked.

What do you mean?
The main difference I see with prusaslicer is the "0" layer for the start gcode.

legend069, 2024-01-18:
During my debugging process with the part cooling PR.
It seems like the support material layers are getting added to the gcode preview slider

TheSlashEffect, 2024-06-30:
I made some progress investigating this. What happens internally in the code is clear, I probably also know why it happens. Looking into more details and posting updates soon.