supermerill/SuperSlicer #4037: Bottom infill generating on top layer

cromine11, 2024-01-02:
### What happened?

When slicing this part, the bottom infill pattern is generating on a specific area of top infill. The 2 settings that affect it seem to be "external Perimeter extrusion width" (setting the spacing doesn't produce this affect) and "infill/perimeter encroachment"
[image]
[image]

### Project file & How to reproduce

[front_body_ECAS_coupler.zip](https://github.com/supermerill/SuperSlicer/files/13809975/front_body_ECAS_coupler.zip)
issue will show on the tap Center

How to replicate:
Print settings -> width & flow -> extrusion width -> external perimeter set 100% (105% does not cause issue and setting spacing to 95% does not cause issue)
Print settings -> width & flow -> overlap -> infill/perimeter encroachment set to 25%

### Version

2.5.59.6

### Operating system

windows 11

### Printer model

Voron Trident

supermerill, 2024-01-05:
You can set your bridge_infill_margin to 0, if you want to limit it.
But yes, a surface that has only one layer and is not in the platter is a bridge and not a top. You have to have 2 layers to have a top.
So you can also split the problematic layers in two with a range layer, so the area can have a bridge and a top. Still, the top quality won't be very precise and the bridge will stick out quite a bit from below.

cromine11, 2024-01-05:
I edited the title to be more accurate. I'm talking about the purple infill in the pictures. My first layer is set to hilbert curve. With certain settings, it generates a first layer in areas where it should be my solid infill (monotonic) or internal bridges.

cromine11, 2024-01-08:
I have another instance the bottom layer generating inside a part in another TAP part. The bottom layer should only generate on the first layer right? The only thing I changed was going from 3 to 4 perimeters but its putting the Hilbert curve in the 4th layer.

3 perimeters:
[image]

4 perimeters:
[image]

infill settings:
[image]

supermerill, 2024-01-08:
ok understood

supermerill, 2024-01-10:
For the last one, it's as expected, imo.
There may be a little gap between the perimeters and the infill, it's useful to put some solid infill behind the perimeter, on top of the notch.
Maybe I can change it from bottom to solid type if the gap is very narrow. But I fear for some unexpected side effects.
To have hilbert only on the platter, you can use a layer height range modifier.

For the bottom in place of the top when the hole shrink and a bottom is detected (because of only one perimeter on top), I think it's safe to say it's a bug and it should stay as top infill.

cromine11, 2024-01-11:
In the last one, the solid infill is not my concern. The lack of an internal bridge is the issue. PrusaSlicer 2.5 also puts solid infill there but it starts with an internal bridge
[image]

To me, its starting to appear like the bottom fill is replacing the internal bridges. The same areas in SS 2.5 I can't fix with by adjusting settings, are the same areas that have internal bridging in PS 2.5
[image]

PS 2.7 does generate internal bridges differently so maybe this could be fixed in that iteration of SS if it can't be fixed now or isn't worth it. If that's the case, I hope this helps with the development at least.

I think the bottom in place of the top infill was just a mix of the same issue exacerbated by other settings because right above that area is 2 more layers.
[image]

I attached the files in the pictures and my SS config.

Problem layers:
**Tap_Front_r8:** 4, 14
**front_body_ECAS_coupler**: 75
**[a]_Tap_Center_r8 No Stop Mod**: 18, 47
[test files.zip](https://github.com/supermerill/SuperSlicer/files/13904074/test.files.zip)

supermerill, 2024-01-29:
I understand.
As the 2.7 modifies a bit the code from here, i'll try to reproduce & fix it on the 2.7, and then move the fix back to 2.5 as early as possible.