supermerill/SuperSlicer #4043: Range printing doesn't work with object instances

vovodroid, 2024-01-03:
### What happened?

PR  **[Merged] Semi-parallel (range) printing #3445** from very beginning didn't support object instances. I recently fix it for Prusa:

```
ObjectsLayerToPrint layers_to_print_range;
...
layers_to_print_range.emplace_back(ltp);
...
this->process_layers(print, tool_ordering, layers_to_print_range,
                                *print_object_instance - object.instances().data(),  smooth_path_cache_global, file);
```
 

### Project file & How to reproduce

Create object instances and slice it with printing range > 0.

supermerill, 2024-01-16:
btw, I was also wondering about making another big change:
Beeing able to print object islands in sequence, ie instead of printing objects after objects after ordering objects: ordering all object's islands and printing the island after islands. So we can have better sorting, and being able to print in range-sequence these (note: with an area threshold to avoid printing thin stuff that may overheat).
More of long-term thinking, i won't do it in the near term.

vovodroid, 2024-01-16:
> i don't know why.

For historical reasons)))

https://intersol.ca/news/organizational-culture-and-the-5-monkeys-experiment/

closed 2024-02-14 by supermerill