supermerill/SuperSlicer #4044: acceleration too fast after last build installation

Jan-Snijder, 2024-01-03:
### What happened?

When installing the newyears upgrade the acceleration is way too high and the Xand Y axis are skipping like hell, almost tearing the printer apart.

And after this slice the printers aceleration stays that high so ther is something changed by this version that sets the machines acceleration way up high.

Check asap please, this is costing us money.

### Project file & How to reproduce

Just load and slice a model

### Version

2.5.59.4 and 2.5.59.6

### Operating system

windows 11

### Printer model

zonestar Z9V5 MK4

SimonQuvang, 2024-01-03:
I am sorry but how is this causing you to lose money? Why wouldn't you just use the previous version and not lose money?

Jan-Snijder, 2024-01-03:
What I ment was its breaking our printer which costs.Not ment to offend but it scared my ass off when I had this experience, man it took my breath away 🥴Sent from Android device

CCS86, 2024-01-07:
This isn't enough information for anyone to solve your issue. 

Compare gcode output of the same project, with different versions, and see what is different.

Jan-Snijder, 2024-01-08:
Hi,

 

No idea what you are talking about ☹ why you di dan acceleration bugfix two versions earlier ? and now the acceleration is bad, way too high, and I must find the difference in the code ? I am not equiped fort hat kind of comparisations with just one printer I like to keep that printer whole and functioning.

 

Sorry but have to wait for another one that reports this or distroys his printer, I won’t go on that road again, sorry.

 

greets

 

Dit e-mailbericht is met Avast-antivirussoftware gecontroleerd op virussen.
www.avast.com

legend069, 2024-01-10:
>No idea what you are talking about ☹ why an acceleration bugfix two versions earlier ? and now the acceleration is bad, way too high, and I must find the difference in the code ? I am not equiped fort hat kind of comparisations with just one printer I like to keep that printer whole and functioning.
> 
>  
> 
> Sorry but have to wait for another one that reports this or distroys his printer, I won’t go on that road again, sorry.
>

You've provided no data that we can use to resolve this "issue".

neophyl, 2024-01-11:
Jan, take your project (a File>Save Project as which saves a .3mf file) and load it in to the 'good' version of the slicer and generate the gcode file as normal that you would send to the printer.
Then take the same project file and load it into the 'bad' version of the slicer, generate the gcode.  Now compare the gcode files (they are just text files).  
If there are acceleration differences then they will be reflected in the gcode.  That will help track down the potential places for the issue.

If you don't have the skills to do that then the absolute minimum required to even start looking at it would be attaching the zipped up saved project file you are using that is causing you the issues and the information of what versions you call good and bad.  That way the devs or other interested parties can load it into the different slicers and do the compare.

There is no need to break anything but unless you can supply the extra info then nothing can be done.

supermerill, 2024-01-11:
@Jan-Snijder
like neophyl said, I can't really know what is "too much".
At the very least I need a project file that create a bad gcode, and to know what are the good version of the software.

Jan-Snijder, 2024-01-11:
Will do as soon as I am ready with printing some stuff I am busy on.

 

greets

legend069, 2024-01-19:
well this is interesting..
writing up the code for the Pressure /linear advance tool https://github.com/supermerill/SuperSlicer/issues/3998 , and i've noticed it too. 
the only place i have "45000" is in printer settings "machine_max_acceleration_x"
i changed that to "60000" and the gcode preview still has "45000" in the lines.
i'm still using 2.4.58.5

weird, wonder how long this has been an issue for. is anyone else able to replicate ?

[image]

[image]

supermerill, 2024-01-22:
project?