supermerill/SuperSlicer #4045: Overhangs undetected

supermerill, 2024-01-03:
### What happened?

Some thin "bridges" in the middle of the part are not detected as overhangs
[image]

### Project file & How to reproduce

[v0_Straight_Cartridge.3mf.zip](https://github.com/supermerill/SuperSlicer/files/13824297/v0_Straight_Cartridge.3mf.zip)

### Version

2.5.59.3

### Operating system

windows 10

Nyshan, 2024-01-04:
This seems to be "fixed" to some degree in the latest version of PrusaSlicer:
[image]

supermerill, 2024-01-05:
it's also not a problem in PS2.5

supermerill, 2024-02-13:
Is the issue still here in the latest nightly?
Seems like it's resolved.

supermerill, 2024-02-14:
understood the issue:
There is something under the bridge to support it in theory, but it's too thin to be printed....
Problem is, I can't know if & when this lower layer has a perimeter or not.
I'll have to grab all settings used by the previous layer to try to compute the minimum width it can handle...
I'll do it in the 2.7, as this area already see numerous changes.

Nyshan, 2024-02-14:
> understood the issue:
> 
> There is something under the bridge to support it in theory, but it's too thin to be printed....
> 
> Problem is, I can't know if & when this lower layer has a perimeter or not.
> 
> I'll have to grab all settings used by the previous layer to try to compute the minimum width it can handle...
> 
> (if it has thin walls activated, then there is something)
> 
> I'll do it in the 2.7, as this area already see numerous changes.

Haha, I couldn't have hoped to have described the problem as eloquently as you've done here. I agree that this is likely what is happening.