supermerill/SuperSlicer #4046: Internal error: Calling AppConfig::save() from a worker thread!  (twice)

sprior, 2024-01-04:
### What happened?

Sometimes when I exit SS I get a dialog box listing:
Internal error: Calling AppConfig::save() from a worker thread! 
Internal error: Calling AppConfig::save() from a worker thread! 

### Project file & How to reproduce

Exit SS

### Version

2.5.59.6

### Operating system

Windows 11

supermerill, 2024-01-15:
Can you replace (saving the previous one) the Slic3r.dll that is in the main directory by this one
[Slic3r.zip](https://github.com/supermerill/SuperSlicer/files/13941257/Slic3r.zip)
and try to reproduce the error?
The error message should be different.

sprior, 2024-01-15:
I tried that dll and it completely lost my environment and opened the configuration wizard.  Fortunately when I returned to the old dll everything came back.

supermerill, 2024-01-16:
yes, it's just compiled with a different internal name (slic3r instead of superslicer)

walkmanyy, 2024-02-06:
I tried new Slic3r.dll, the  error message changes to: Internal error: calling GUI APP:: post_init() from a worker thread!

closed 2024-02-14 by supermerill