supermerill/SuperSlicer #4048: [BUG] M204 P0 T0 in RepRapFirmware

timschneider, 2024-01-04:
### What happened?

The slicer will output M204 P0 T0 in G-Code causing the printjob to fail.

### Project file & How to reproduce

Change default acceleration under print settings -> speed -> Default acceleration to a value != 0 -> slice -> change it back to 0 -> slice again -> M204 appears with P0 T0

https://github.com/supermerill/SuperSlicer/assets/2517553/b9c118b5-7f05-4164-b09e-ff0ff49a56ed

[Shape-Box.zip](https://github.com/supermerill/SuperSlicer/files/13830027/Shape-Box.zip)

### Version

2.5.59.6

### Operating system

Windows 11

### Printer model

Meltingplot MBL 136

vknvkz, 2024-01-08:
I have the same problem but with Gcode Flavor = Marlin 2.
This problem appeared in version 2.5.59.5

vknvkz, 2024-01-08:
I found a solution, all acceleration values in Print Settings -> Speed -> Acceleration control (advanced) must be 0.
Even the grayed-out cells.

timschneider, 2024-01-08:
> I found a solution, all acceleration values in Print Settings -> Speed -> Acceleration control (advanced) must be 0. Even the grayed-out cells.

I would call it a work around :) 
If the values are greyed out, they should be ignored by the software and/or set to a value that does not cause any damage.

supermerill, 2024-01-10:
thanks for he report.
I made 2 errors:
 . forgot to initialize the travel acceleration to 0 at startup
 . I had a safeguard for 0 travel accel, but not yet for 0 printing accel.

closed 2024-02-14 by supermerill