supermerill/SuperSlicer #4049: Individual Object aka Klipper "EXCLUDE_OBJECT" errors

supermerill, 2024-01-04:
### Discussed in https://github.com/supermerill/SuperSlicer/discussions/4040

<div type='discussions-op-text'>

<sup>Originally posted by **tocguy** January  3, 2024</sup>
I am unexpectedly getting errors appearing in Octoprint's Klipper terminal in newly-sliced gcode files. They relate to the appearance of unknown "EXCLUDE_OBJECT" commands in the gcode file. I double-checked Superslicer's settings and the **_Complete individual objects_** checkbox is cleared. I toggled it ON and OFF again to no avail. The unwanted "EXCLUDE_OBJECT" commands remain. The gcode file contained only a single object anyway. With multi-object gcode files the problem is compounded. Is this a corruption in my slicer config? Is there something about Superslicer that I'm simply unaware about?

I re-sliced using the default slicer config and the issue did NOT repeat soooooo yeah, somehow Superslicer got a bee in its bonnet about  **_Complete individual objects_** .

Any idea why or how to discover what may be the cause?</div>

Lemcott, 2024-01-04:
Unselect "label items"

Or add [exclude_object] to your printer.cfg

Not a susi issue.

tocguy, 2024-01-05:
TY Lemcott,
It was due to Superslicer being updated to include the EXCLUDE_OBJECT capability but my version of Klipper was not there yet. Updated Klipper and added "[exclude_object]" to Klipper's printer.cfg. All is good now. As you stated, simply unselecting "label items" in the slicer would also work.