supermerill/SuperSlicer #4051: vase mode creates gap in print

trampas, 2024-01-04:
### What happened?

When using vase mode and slicing a model the slicer will often create a gap in the print. 
[image]
[production.zip](https://github.com/supermerill/SuperSlicer/files/13833330/production.zip)

### Project file & How to reproduce

Open the project and slice the enclosed model note in gcode view that around 23.20 into print the perimeter is missing. 

I suspect that the vase mode could not slice it correctly, ie not a loop due to layer height issues and precision.  However the slicer should at least warn that this happened. 

### Version

2.5.59

### Operating system

windows 11

### Printer model

voron 2.4 with stealthburn

Lemcott, 2024-01-04:
I believe this is a latent PS bug that has been fixed since 2.5. I'm not 100% certain on that, would need to verify

supermerill, 2024-02-29:
Prusa did stuff on the issue, I saw it when merging the 2.7