supermerill/SuperSlicer #4052: Show actual speeds

trampas, 2024-01-04:
When setting speeds in super slicer I often wonder what actual speeds will be, for example when setting the "brim speed" it would be nice if next to the brim speed there was a box showing the current brim speed based on all the settings.  Then if you click in this box it would tell you what parameters were controlling that speed.

supermerill, 2024-01-04:
if you put a %, having the raw value alongside?

trampas, 2024-01-04:
Yes ! Have the actual value being used, and in the pop up tech tip have the parameters and equations that lead to that value 

For example on the bridge speed beside the percentage (or mm/s) it would have grayed out box showing the actual mm/s. If the bridge speed is a percentage then the tech tip would say something like "default speed is 100mm/s, 50% of that is 50mm/s for bridge speed."