supermerill/SuperSlicer #4053: Layer time goal not working

trampas, 2024-01-04:
### What happened?

The layer time goal setting in "filament settings" are not working. 
[image]
[image]

The images above show that the layer time goal should be 100s 
[image]

The speed never reaches 5mm/s as requested 

### Project file & How to reproduce

Open the zip file and project and give it a try yourself 
[elevators_2hr.zip](https://github.com/supermerill/SuperSlicer/files/13834321/elevators_2hr.zip)

### Version

2.5.59

### Operating system

windows 11

### Printer model

voron 2.4 with stealthburn

supermerill, 2024-01-04:
can't reproduce on 2.5.59.6
if you go to help -> about, what is your version?

trampas, 2024-01-04:
It is 2.5.59.

It appears to work in prusaslicer 2.7.x

I assume it is a PS bug that was fixed. Also parameters for the feature is different in PS 2.7.

Murrdo, 2024-01-05:
At the moment, the bug is fixed in 2.5.59 (Win10 - the latest nightly build), I've been waiting for it to be fixed for a very long time)

supermerill, 2024-01-05:
there was also a bug in 59.4 and prior, fixed in .5 and .6
https://github.com/supermerill/SuperSlicer/issues/4015
that's why i want to know your full 4-digit version

trampas, 2024-01-05:
Mine is 2.5.59.3

ercdvs, 2024-01-17:
I am on 2.5.59.6 and I still have this issue.  In my understanding,  The top most layer should be 15s, but are currently 1s

[image]

[image]

[image]

supermerill, 2024-01-17:
@ercdvs i need your project file to test myself, to reproduce and fix.

ercdvs, 2024-01-17:
> @ercdvs i need your project file to test myself, to reproduce and fix.

Added to my original comment

supermerill, 2024-01-17:
On 2.5.59.6
[image]
the only parts where it goes below 15sec is when it's at the very top.

supermerill, 2024-01-17:
use `layer time (log)` (the second one) to have a better scale

Nerobeus, 2024-01-23:
I found away to reproduce, it happen when changing any speed for print move in print settings and can't be reverted without loading a previous saved project or delete all profiles.

closed 2024-02-11 by supermerill