supermerill/SuperSlicer #4054: retract at layer change with avoid crossing perimeter

CuredPrusa, 2024-01-04:
### What happened?

When <avoid_crossing_perimeters> is enabled together with <retract_layer_change>, slicer is unnecessary avoiding perimeters at last travel move towards the seam, although the filament  is retracted. 
I expect the straight travel towards the seam, which saves time and preserves nozzle pressure. 

### Project file & How to reproduce

Slice any object with internal perimeter with mentioned options enabled. 

### Version

2.4.58.4

### Operating system

W10

### Printer model

V400 / Klipper

supermerill, 2024-01-05:
I think it's better to make the travel, with avoidance and then do the layer change, unless it cross a perimeter, then I agree.

CuredPrusa, 2024-02-13:
The problem is when it travels a lot around perimeters when reaching the seam for another layer. It loses pressure and seam can not start right. I do not retract much, therefore I avoid crossing perimeters. But for the sake of a good seam, I retract at layer change and nozzle should be at new seam position asap.