supermerill/SuperSlicer #4055: Tech tips go away

trampas, 2024-01-04:
### What happened?

When you change a parameter, for example "bridge speed" the pop up tech tips, aka help, stops working until you click on another screen and go back to previous screen.

### Project file & How to reproduce

Change parameter and see if tech tips still work 

### Version

2.5.59

### Operating system

windows 11

### Printer model

voron 2.4 with stealthburn

supermerill, 2024-01-05:
I don't really understand.
What I tried:
 . I activated tech tips.
 . I started the slicer
 . a tech tips is displayed
 . I went to a preference tab, and changing a setting
 . I went back to the platter
 . The tech tips was still here
 . I tried to click on the 'X' button to close it, it works.

Maybe because I'm still on win10

trampas, 2024-01-05:
So I go to setting the tech tip pops up:
[image: image.png]

I change the value:
[image: image.png]
Then mouse over min length and nothing happens.

I think the issue is that the tech tips stop if the cursor is still in the
parameter setting, aka text box entry is active.  That is you have to hit
enter or exit screen/tab for the parameter to change and tech tips to
reappear as the 'focus' is still on the parameter being changed.

A possible way to fix this is when a text entry box is active, highlighting
it in red, or changing background color. This would inform dumb users like
me that they are still active entering data in the field, i.e. have not hit
enter.

Thanks
Trampas

closed 2024-02-14 by supermerill