supermerill/SuperSlicer #4056: Move "infil/perimeters encroachment"  into plastic profile.

Murrdo, 2024-01-05:
The "infil/perimeters encroachment" setting in the print settings section, in my experience, is very individual for each plastic. I have plastics that require values ​​from 4.5% to 9% and this greatly affects the quality of the top layer. ("ridges" at the junction of the filling and perimeters on the top layer)
And this setting is in no way compensated by the flow.
At the moment, it is very inconvenient to write down this value in notes in the plastic profile and change it in the print settings section before each print. Is it possible to transfer it to the plastic profile?

supermerill, 2024-01-05:
It's possible.
But I need more than one report.
There may be people that prefer it on the print section, or maybe they all agree with you.
Imo, why not, seems sensible.

shinanca, 2024-01-05:
It`s really need. Best approach stay settings in Print Settings, but can be overwritten by plastic.

PetrSin, 2024-01-06:
Absolutely agree! Customizing the "infil/perimeters encroachment" setting for different plastics is crucial for achieving optimal print quality. Transferring this setting to the plastic profile would indeed be a helpful improvement, streamlining the process and saving time before each print. Your suggestion could significantly enhance user experience in managing diverse printing materials.