supermerill/SuperSlicer #4057: Advance Klipper support

trampas, 2024-01-05:
I have moved to klipper and will never go back, sure not everyone will do this, but I feel klipper is the future. 

On that note the "printer settings"  it would be nice if we could download them from printer it's self.  That is the klipper configuration file(s) contain most if not all the printer settings.  As such when I select the printer it would be nice to have option to set "printer settings" from klippper directly. 

In an ideal world I would have something like ERCF and/or multiple tool heads (tool changer) on the printer. Then the tool head (extruder), and the filament settings would all be in config files on the printer.  So for example I would select the printer, and which filament I will be using for a print.  Then SuperSlicer would download the config data from the printer and slice the model with correct settings for filament used, printer used, and tool head used.   
This becomes increasingly powerful when you get to print farms with various printers in the farm.  That is you can take the same STLs  and do slicing for each printer and be confident you are getting quality results.   

Eventually I would see superslicer being compiled using web assembly (webasm?) such that it runs in browser.  Then using Klipper you would just go to printer's local website (mainsail for example) which would host superslicer, upload your STL and slice for that printer all in the browser.  No more installs, easy software updates for bug fixes, etc. 

At this point superslicer could also generate Klipper specific GCode with macros and other advance features, like phyton in gcode.   For example when you do a "Z hop" you might have code like:
G1 E-0.75 F20 ;retract filament
G1 Z0.2 F100 ; raise the nozzle
These could be replaced with variables in Klipper something like:
G1 E[zhop_retract_dist] F[zhop_retract_rate]
G1 Z[zhop_height] F[zhop_zfeed_rate] 
These variable would be set at the start of the GCode and now the operator on the fly could change the variables as the print runs.  Yes klipper would need to support this as well as web interfaces like mainsail. 

Of course most operators will most likely never change the variables, but it also allows dynamic control of prints, for example adding cameras and AI in the future then it could tweak parameters on the fly based on sensor feedback. 
This idea is much like z offset or "baby steps" generalized to create gcode variables.  Which we do currently via operator in the loop, but eventually we will have computer vision and AI where it could change print parameters "on the fly". 
Again this would require some collaboration between SS and klipper developers to make this happen.  However it would also greatly increase value of SS.  

The real power of this for SuperSlicer is that once these features are done via web assembly, ie native in browser,  then OEM printer manufactures will want this.  Because they can contract developers of superslicer to make vendor specific features and changes.  This allows the OEM manufactures to differentiate their products by making it easier for their customers.  They can basically require no software install on PCs to use printers.  They then can start working on advance real time print quality sensors and have means to dynamically control printer.   Additionally by having the online updates that klipper does currently it allows them to push slicer and printer changes quickly and improve the quality and value for their customers.   

The other big customer is print farms. That is having a heterogenous set of printers running klipper they could farm out work to printers and have them custom slice models for that printer.  Here they would need a web interface that monitors printers, and when you upload STLs it will recommend printer to be used based on print speed, print cost (machine hourly cost), filament availability on the machine, and quality of print.   This print farm management interface is another product that could be developed and sold. 

The world is changing and the discrete independent slicer is going away, the integration between controller (Klipper) and slicer will be married and coexist with a web browser interface.

dougshell, 2024-01-05:
Curious, where would the resources for this web-based slicer come from?

trampas, 2024-01-05:
I have not done it yet, but as I understand it, compiling for web
assembly is very similar to compiling for another targeted processor.
https://webassembly.org/getting-started/developers-guide/
It is on my list of things to learn to do.

The idea is to keep the same code base and just port to web assembly.

linux-paul, 2024-01-08:
It's a double-edged sword for me. Klipper macros are evaluated line by line, which leads to exciting effects with nesting. Here you would have to ensure that automatically generated macros do not violate any existing structures. In my case, I don't use Z Hop at all with pressure advance (see the documentation 'It is recommended to disable the slicer's "z-lift on retract" option'), but I do use the FW retract. -> I don't think an implementation will be easy.

supermerill, 2024-01-16:
> On that note the "printer settings" it would be nice if we could download them from printer it's self. That is the klipper configuration file(s) contain most if not all the printer settings. As such [...]

importing the klipper printer.cfg to fill a printer profile? I guess I can do it in the wizzard... It's not that big of a deal if it can be parsed by boost.

> In an ideal world [...]

We'll go step by step

> that it runs in browser

not interested myself.
But I will make an api so other web interface can be the gui for superslicer.
And a tab for displaying the firmware interface (if you want the opposite) like octoprint.

> At this point superslicer could also generate Klipper specific GCode with macros and other advance features, like phyton in gcode. For example when you do a "Z hop" you might have code like:

In the far future, it still planned to have multiple gcode writer/reader as external module, so these kind of thing are possible.