supermerill/SuperSlicer #4061: Artifacts due to multiple causes

Murrdo, 2024-01-08:
I apologize, I know that this defect has already been described in issue #3785. But it is marked as closed, I left comments in it, but no response.
It also seems to me that I have found more reasons for the appearance of artifacts.

Version: 2.5.59.6
Build: SuperSlicer_2.5.59.6_2024-01-05
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19045), 64-bit edition

I found several reasons that provoke the appearance of perimeters hanging in the air:
1.Often the “wall thickness” parameter spontaneously takes on a value with 4 decimal places
[image]
[image]
If you change the value manually, removing the decimal places, there are no artifacts.
Also, some extrusion width values ​​lead to the same effect, but I did not understand the pattern
[One artefact.zip](https://github.com/supermerill/SuperSlicer/files/13855822/One.artefact.zip)

2.When the generation of arches (G2/G3) and "threshelod for overhangs" with arachne is turned on at the same time, a huge number of artifacts appear.
[image]
[Many Artefacts.zip](https://github.com/supermerill/SuperSlicer/files/13855823/Many.Artefacts.zip)

Also, if you place this model in the center of the table in the default position, the slicer crashes with an error on layer 74.
[image]
[Error.zip](https://github.com/supermerill/SuperSlicer/files/13855835/Error.zip)

TheBoojah, 2024-01-31:
I also found that setting "Model rounding precision" to 0 (default 0.0001) removes a lot or all of the hanging perimeters.

supermerill, 2024-02-01:
The issue was that the gcode reader wasn't refreshing its position from G2 move, and so the fan mover infer the wrong position, hence the lines going anywhere.
But if there isn't a G2 just before a G1 (both are G2 or both are G1, or it's G1->G2) where the fan is moved, then there is no artifact.
The gcode reader is fixed in the current nightly & next releases.

closed 2024-02-14 by supermerill