supermerill/SuperSlicer #4064: make 100% sparse infill solid infill

supermerill, 2024-01-08:
Currently a 100% sparse infill is still not a solid infill.
If their pattern is the same, then it should be considered as solid infill.

Nbeknel, 2024-01-17:
Regarding speeds, will there be made a difference between "standard" solid infill and solid infill where normaly sparse infill would've gone? OrcaSlicer has currently this feature enabled and a lot of the times I've noticed a significant increase in print time when printing with 100% infill compared to Prusa/Susi with similar settings or even compared to itself with 99% infill.

Orca(100%) - 16h 47m
[image]

Orca(99%) - 15h 29m
[image]

PS - 14h 49m
[image]

SuSl - 13h 53m
[image]

Projects:
[Benchy_300%.zip](https://github.com/supermerill/SuperSlicer/files/13968055/Benchy_300.zip)

supermerill, 2024-01-18:
I think with 99%, it uses the sparse infill speed, and 100% the solid infill speed.
But is it really an issue?

Nbeknel, 2024-01-18:
I believe in some cases it could be problematic. If we look at it from the perspective of throughput, this change could lead to a decrease in production, when previosly you could print x instances per hour, now you would print y<x instances per hour.