supermerill/SuperSlicer #4067: Empty Layer Detected due to Support

tocguy, 2024-01-11:
### What happened?

SS: 2.5.59.6
Windows: 10 22H2

Been getting some strange behavior when slicing. Occasionally, an "Empty Layer Detected" error appears. I use Solidworks to develop the models and then re-import them and use SWs Mesh Diagnostics to check for errors. The attached file generated the error in SS. 
[image]

I ran Netfabb on it and it reported that it corrected the file but alas no change to the slicing. Same error.

Next, sliced the STL in Cura 5.4 without error. Hmm...

Switched back to SS and toggled Support OFF completely. Sliced again. No error generated. OK, so toggled Support ON with Support on Build Plate Only selected. This time no error.

Go figure!?!?!?

FWIW, the empty layer was detected between 4.1 and 4.8 which when using the gcode slider revealed that only a tiny bit of support was being placed and zero elsewhere.

I'm interested in others results as this appears to be a recurring scenario both with SS and Prusa although admittedly it is frequently the model that is at the root but for the error to just "go away" like it did by toggling Support is kinda weird.

### Project file & How to reproduce

[Web.zip](https://github.com/supermerill/SuperSlicer/files/13904355/Web.zip)

### Version

2.5.59.6

### Operating system

Windows 10 22H2

### Printer model

Anycubic MegaZero 2 - an Ender 3 clone

supermerill, 2024-01-11:
I tried to slice your project, and there is no issue for me.
I tried to remove and re-add the stl, same.
Can you tell me how to reproduce from a blank state?

tocguy, 2024-01-11:
Hmm, given that the 3mf project already contained the settings the only thing I did was import the stl, use the object tool to pick which side to lay on the bed, then slice.
Tried it again and the error is reproducible. The issue is isolated down to toggling Support from Everywhere to Build-Plate Only.

supermerill, 2024-01-11:
ok, i had to change the support to "everywhere"

supermerill, 2024-01-29:
Your piece was very useful to debug some other minors things.
But now I can't reproduce the bug again. Maybe i fixed it in the meantime?
Can you try to reproduce with the latest [nightly build](https://github.com/supermerill/SuperSlicer/actions/runs/7680530505/artifacts/1199927626)?

tocguy, 2024-01-29:
Grabbed the nightly_win64 commit 4b8ded3 and incurred no error while re-slicing the object with support Everywhere.TY

tocguy, 2024-01-29:
Outbox error so resending.