supermerill/SuperSlicer #4071: Crashes after a minute or two

Naomitr0n, 2024-01-14:
### What happened?

SuperSlicer opens, then after a minute or two the taskbar icon goes red, and after a few seconds it crashes. It does this whether I'm quick enough to slice a print or if I just open it and let it sit there. It doesn't do this on my desktop but does on my Acer Swift 3 laptop. I tried copying the opengl32.dll into the main directory but no dice. Keeps happening on 2.4.58.5, but I tried out 2.5.59.6 to see if it made any difference and it crashes on that version too, whether the opengl32.dll file is in the main directory or not.

Doesn't seem related to anything other than the time it's open. I can have nothing else open, or a lot of stuff open, I can be plugged in, or running on battery. My only hunch is that it might be related to the 200% scaling I have due to it being a 14" 2880x1800 display? Or HDR or something? I really can't tell. 16 GB of RAM should be plenty with nothing else open, even if I use a couple custom bed models (though the one I'm mostly using is just a flat bed with a texture, nothing else around it). Attached a couple .dmp crashdump files in the ZIP along with an empty profile, hoping this issue isn't too complex to sort out.

### Project file & How to reproduce

[SuperSlicer Bug Report.zip](https://github.com/supermerill/SuperSlicer/files/13929972/SuperSlicer.Bug.Report.zip)

### Version

2.4.58.5, 2.5.59..6

### Operating system

Win11

### Printer model

M3D Crane, CR-10 V2

supermerill, 2024-01-15:
Can you copy your SuSi and extract this zip inside (should replace the existing files) ?

[Debug.zip](https://github.com/supermerill/SuperSlicer/files/13938708/Debug.zip)

Then try to run it (from command prompt: win+r, 'cmd' then enter, drag& drop the superslicer_console.exe then enter) , make it crash, and give me what is in the comamnd prompt, the crahs log, etc.
It should create a better crash log, as it's compiled with debug information.

Naomitr0n, 2024-01-15:
> Can you copy your SuSi and extract this zip inside (should replace the existing files) ?
>
> [Debug.zip](https://github.com/supermerill/SuperSlicer/files/13938708/Debug.zip)
>
> Then try to run it (from [...]

Did that, but I'm getting some errors when it tries to run. Missing a couple DLL's it seems like:

> The code execution cannot proceed because MSVCP140D.dll was not found

That happens once, this one happens three times in a row:

> The code execution cannot proceed because ucrtbased.dll was not found

Can you ZIP those up too for me? Don't want to chase down the wrong version and make this a "DLL Hell" debug. I don't see them in the base directory and I didn't see them in the repo either.

Side note: how have I been alive 33 years and not known about the "drag-drop into command prompt" trick? Works in ConEmu too, that's wild, thanks for that! :)

supermerill, 2024-01-16:
oh, yes, he want debug dll.
These should cover you, i guess.
[msvcp140d.zip](https://github.com/supermerill/SuperSlicer/files/13954032/msvcp140d.zip)

Naomitr0n, 2024-01-16:
Still missing `ucrtbased.dll`, can you include that one too? Crashes trying to find it

supermerill, 2024-01-17:
[ucrtbased.zip](https://github.com/supermerill/SuperSlicer/files/13965639/ucrtbased.zip)

Naomitr0n, 2024-01-17:
Symptoms seem to change depending on whether I'm on battery power or plugged in. I was on battery power when it happened constantly, foreground or background, but plugged in it only crashes if I'm in the foreground trying to slice something and change settings -- Took me three tries to get a screenshot of the error message, so I included all three crash dumps. I had sliced the model and was changing the infill by typing it each time, didn't leave the main screen.

Win11 settings for `System > System & Battery > Power Mode` is `Best Performance` when plugged in, but `Best Power Efficiency` when on battery power -- might be making a difference?

Uploading also the `Closet Door Spacer.3mf` file I used to get it to crash when plugged in (which I didn't need when on battery power). Literally just a cylinder spacer I made in Fusion360 in about a minute, nothing special. Not sure if it's relevant, but I'm also using 3D bed models in the preview so I've included those too. I'm slicing it under the `Robobitch` profile so it's the lighter weight of the two previews that's rendering literally just the bed surface, nothing around it.

[2024-01-17.zip](https://github.com/supermerill/SuperSlicer/files/13969088/2024-01-17.zip)
[Closet Door Spacer.zip](https://github.com/supermerill/SuperSlicer/files/13969091/Closet.Door.Spacer.zip)
[Custom Bed Models.zip](https://github.com/supermerill/SuperSlicer/files/13969127/Custom.Bed.Models.zip)

supermerill, 2024-01-18:
For the error message, go in preference -> gui and check the "hide tooltips on on slice button"

Is these crash with a fresh new configuration? (like when you create a new configuration directory next to the exe)
If not, you can zip me your configuration directory, maybe something inside makes the bug appear.

Naomitr0n, 2024-01-18:
Here's the config I was using for 2.4.58.5, definitely not fresh by any means, I made it originally on my gaming rig and exported-imported it onto my laptop :)

[configuration.zip](https://github.com/supermerill/SuperSlicer/files/13979883/configuration.zip)

I realized I was plugged in when trying to see if changing that setting affected things while I was on battery power, so I unplugged the laptop, and immediately it hung and crashed. Extremely repeatable too:

1 - Open SuperSlicer when plugged in
2 - Open 3MF file (this might be optional -- used the `Closet Door Spacer.3mf` from above)
3 - Unplug laptop
4 - SuperSlicer immediately hangs for a bit before crashing

I imagine this is the same crash as when it's already on battery power but sitting in the background, but not fully sure on that. I opened up the debug version of 2.5.59..6 we're working with and got a couple crash logs from one crash of SuperSlicer going through the above 4 steps:

[2024-01-18 - Crash Dumps When Unplugging.zip](https://github.com/supermerill/SuperSlicer/files/13980041/2024-01-18.-.Crash.Dumps.When.Unplugging.zip)

Console read the same as the other crash dumps, just something about "Unable to parse version of AMD driver." I'm not sure why it's trying to grab AMD versions for Intel CPU/GPU setups though -- [here's the laptop specs](https://www.acer.com/us-en/laptops/swift/swift-3-oled/pdp/NX.KAVAA.001). Wondering if this is some kind of Win11 laptop-specific power management bug given how it seems to happen when the OS would have a reason to try and conserve resources while on battery power.

supermerill, 2024-01-18:
Problem is, the dcrash dump is still not very useful, as it just say "i crash with error X". It should be able to say where in the code it was when crashing.
I made some research, and it seems windows need to have a registry key changed from 1 to 2 to do that.
documentation: https://learn.microsoft.com/windows/win32/wer/collecting-user-mode-dumps
1- open regedit (win+r -> regedit)
2- go to `HKEY_LOCAL_MACHINE\SOFTWARE\Microsoft\Windows\Windows Error Reporting\LocalDumps`
3- change the value of the key DumpType from 1 to 2
4- reproduce the crash and resend me the dump. Hoipefully it will be better.

For the AMD 'error' message on the command prompt, you can ignore it, it's just a check.
 
I tried with your configuration, but I really can't reproduce it on my amd laptop (win10), connected to power or not.
I may need to buy the same laptop model as you to reproduce the error, I hope not.

Naomitr0n, 2024-01-19:
I'm with you on the hoping you don't have to shell out a grand for this laptop. Definitely will do what I can to help you remotely diagnose what's going on! :)

Got three crash dump groups for ya. The `Unplugging` ones I opened SuperSlicer and unplugged the laptop, got it to crash. I opened SuperSlicer again, it crashed *immediately*, so those are in the `Battery Immediate` folder. Then replicated the bug we've had before and put those in `Battery In Background` folder. I've been going back to Google Chrome and looking at a bunch of tabs and waiting for it to hang and crash each time for those. Part of me wonders if the 16 GB of RAM is playing a role somehow, but every other app I use seems to handle it fine so I'm really not sure.

[2024-01-19.zip](https://github.com/supermerill/SuperSlicer/files/13993053/2024-01-19.zip)

Hope these crash dump logs do the trick! :)

legend069, 2024-02-10:
do you still get the crashes in 2.5.59.7 ?

Naomitr0n, 2024-02-10:
> do you still get the crashes in 2.5.59.7 ?

Unfortunately yes, just tested it, still having the same issue. I'm not able to get debug crash dumps of the new version, looks like the files I replace make the splash screen read as if it's still 2.5.59.6, so I didn't want to mess with that approach. Fresh ZIP download of 2.5.59.7, opening up the `Closet Door Spacer.3mf` file, and unplugging the laptop makes it crash immediately. So does being on battery power, loading up the `Closet Door Spacer.3mf` file, slicing it, then heading to print settings and then making any random Google Chrome window the focus and waiting for about a minute. No change in behavior.

I don't know if it was crashing for me when the laptop was plugged in other than the potential "tooltips on slice button" setting being in the wrong spot as supermerill mentioned five comments above that's now fixed, which has me wondering if it has to do with some kind of Win11 battery management thing. My theory is that Win11 is trying to put SuperSlicer in the background in a way that makes SuperSlicer crash. Not sure why SuperSlicer is crashing if that's the case however.

supermerill, 2024-02-11:
Sadely, I can't really fix it.
Idealy, I need to compile it on your computer in a debug build, and running with the debugger to see what's going on.

btw, your zip is corrupted.

Naomitr0n, 2024-02-11:
> Sadely, I can't really fix it. Idealy, I need to compile it on your computer in a debug build, and running with the debugger to see what's going on.
> 
> btw, your zip is corrupted.

Which ZIP got corrupted? I'll re-upload it for ya.

Gonna rename the issue and tweak the first post to make it a lot clearer so others can find it and hopefully chime in if they're also having the issue on cheaper hardware. Wouldn't hurt to put something in the release notes asking if anyone is having the same problem on a laptop that costs significantly less than a thousand dollars and can share what hardware they're running. With luck it'll fix it on my laptop too :)

For now, leaving it plugged in avoids any/all crashes now that tooltips on the "slice now" button are disabled, so as long as I'm near an outlet I'm good.

neophyl, 2024-02-11:
Wonder if windows is doing something with the graphics drivers/card when going to battery mode ?  If you start SuSi up using the software rendering option you could try that on power and battery and see if it makes any difference.  Another data point at least.

supermerill, 2024-02-12:
> Which ZIP got corrupted? I'll re-upload it for ya.

the last one, 2024-01-19

Naomitr0n, 2024-02-13:
Tried to find the crash dumps, must've gotten rid of them, so I tried to make new ones. Then I, uh... couldn't.

I don't know what I did but something's suddenly working as intended in the debug build, but if I switch to the latest version, it's still broken. I don't even need to open a 3MF file between the two to replicate the results. Problem is, it uses basically ALL of the CPU for about five minutes to slice a simple cylinder like the `Closet Door Spacer.3mf` I attached above. The `--sw-renderer` option doesn't seem to make a difference in the time to slice, CPU usage, nor whether or not it crashes. The console gives additional messages about loading the OpenGL driver for the `--sw-renderer` option but that's about it.

Weirdly the debug build now has an error message saying:
> No gcode_ascii in ConfigOptionsGroup config, tab printer_fff.ui.

GitHub attachment limit is 25MB and it doesn't like multi-part files, so I threw the multi-part 7Zip file on Discord since their CDN is usually okay with hotlinking. Let me know if the links are dead or if there's a better way to upload larger files though.

This is the debug build that works, using the files you provided:
[SuperSlicer_-_DEBUG.7z.001](https://cdn.discordapp.com/attachments/767176029063479318/1206989215217487993/SuperSlicer_-_DEBUG.7z.001?ex=65de034d&is=65cb8e4d&hm=0ed0f65905e440409e99d11bd65f0e9f922a057d0fdd6b2c854f05e3f8434684&)
[SuperSlicer_-_DEBUG.7z.002](https://cdn.discordapp.com/attachments/767176029063479318/1206989215695503370/SuperSlicer_-_DEBUG.7z.002?ex=65de034d&is=65cb8e4d&hm=909a81ca9fbad5b1b20c8769496c3b1e08a4280d7dcafe8cdac86cbb3129c157&)

This one doesn't:
[SuperSlicer_2.5.59.7.7z.001](https://cdn.discordapp.com/attachments/767176029063479318/1206989216064606329/SuperSlicer_2.5.59.7.7z.001?ex=65de034d&is=65cb8e4d&hm=450735dcf314895ad2c1e84c7b33f23c630a1507d43834aa5dcbed62759cc2ed&)
[SuperSlicer_2.5.59.7.7z.002](https://cdn.discordapp.com/attachments/767176029063479318/1206989216442097734/SuperSlicer_2.5.59.7.7z.002?ex=65de034d&is=65cb8e4d&hm=9065c57d15901c3674a3bf9defbbe2c0dc490a43afc30a1962dd084cd74a810e&)

So we have an example that works but overloads the CPU, and one that doesn't and barely touches the CPU. Not sure what to make of that.

supermerill, 2024-02-13:
The `SuperSlicer_2.5.59.7.7z.001` can slice a simple cylinder instantly without crash on my computer.
Not sure how I can debug it.
Maybe I can connect to your pc with a remote connection, install what's needed for debug and try debugging?

Naomitr0n, 2024-02-13:
> The `SuperSlicer_2.5.59.7.7z.001` can slice a simple cylinder instantly without crash on my computer. Not sure how I can debug it. Maybe I can connect to your pc with a remote connection, install what's needed for debug and try debugging?

We could try that, but I'd need to be physically present so I can unplug the laptop when you're ready for me to do so since it doesn't have issues when plugged in. Let me know how we can set that up, and if I can install anything needed for debugging ahead of time :+1:

Jameson42, 2024-04-30:
Chiming in here, I also have this same crashing issue when on battery on my Acer Predator Helios running Windows 11. However, if I run SuperSlicer as admin the issue does not occur.

Naomitr0n, 2024-04-30:
> if I run SuperSlicer as admin the issue does not occur.

Checked both my 2.4.58.5 install and a brand new 2.5.59.9 download as well. In both cases: open SuperSlicer, unplug laptop, instant hang and crash. I do remember trying running as admin on the 2.4.58.5 install but wasn't sure if I did on the 2.5.x branch, but guess it doesn't work for either on my end.

I was hoping for an easy answer but the best answer I have is "don't unplug the laptop when you're using SuperSlicer" which for my use case isn't too terrible, it just limits mobility a bit.

Interesting that they're both Acer though, wonder if there's anything to that

supermerill, 2024-05-15:
Do you have a dedicated graphic card?

supermerill, 2024-05-15:
I think yes.
Maybe it's the laptop, switching from dedicated graphic to internal one, and the opengl driver / the slicer opengl code freak out and crash.

what you can try:
 * force superslicer to use the internal graphic card.
 * force the internal graphic card as default even when power is supplied, only use the dedicated GC for when you launch a game.

Naomitr0n, 2024-05-15:
> Do you have a dedicated graphic card?
I don't think my Acer Swift 3 does, at least not that I can tell. Is it possible the integrated graphics on mine is using some behind the scenes stuff to switch between foreground and background modes though and causing issues? Let me know what I can try to diagnose further.

Naomitr0n, 2024-08-23:
Tried out the newest version, it is still having the "unplug, immediate hang/crash" problem, but it's also hanging and crashing left and right on it's own even without any files opened, plugged in or not. It's kinda nuts I have about 30 seconds to get in, change things, then save the 3MF before it dies again.

It seems to be amplified by low available RAM -- 16GB doesn't leave a whole lot of room for other stuff on a laptop -- but even with nothing else running it's just killing itself left and right. Just open it, click around in the menus and such, and within 2-3 minutes it's hung and crashed. Foreground entire time too.

Is there a way to have the new version generate crash logs too? It's still complaining about not being able to figure out the AMD graphics driver version despite being Intel graphics for some reason as an "error", but it keeps running after that error message is thrown.

legend069, 2025-05-02:
do you still get this unknown crash with 2.7?

Naomitr0n, 2025-05-03:
> do you still get this unknown crash with 2.7?

Just tried the latest version as well as the 2.5.60.0, here's my results:

2.5.60.0
- immediate hang/crash when unplugged
- hangs/crashes quickly if opened on battery power and sent to background

2.7,  2.7.61.1
- [untested]

2.7.61.2
- unplugging laptop does not cause crash
- plugging laptop back in does not cause crash
- leaving "blank" SuperSlicer open in background does not crash
- opening a 3MF project and putting that in the background after some minor changes does not crash
- Opening an STL and putting that in the background also does not cause crash

Not sure what changed between the versions that's relevant to this crash, but it does appear to be fixed, so I'll mark this as closed :)

closed 2025-05-03 by Naomitr0n