supermerill/SuperSlicer #4075: Add confiramtion box for color change

supermerill, 2024-01-16:
It's easy to click at the wrong place and add a color change and so invalidating the current sliced code.
It's best to add a preference item so a confirmation dialog prevent the color change if it was unintentional.

supermerill, 2024-01-23:
For miss click
Often, I just want to look to another layer and I add a color change and then the gcode is invalidated and i have to wait for it to recompute before removing it.
One day, i'll do something to keep the invalided data, and color the slice button red instead, or something.

legend069, 2024-01-31:
i thought about approaching this by just moving the box for it a few pixels over, or making it larger/smaller.
but the ``don't show again`` bit for it would prob be best.

if adding a "are you sure" checkbox, it should come with a ``enable/disable, do not show again`` 
just to save the possibility it could annoy users that utilize the multi-material features frequently.

supermerill, 2024-01-31:
yes

supermerill, 2024-01-31:
You're right, there is enough space.
I pushed the button to the far side
Also added a little empty unresponsive space between the lock button and the bottom of the slider, as I also do many miss clicks here.
I'll see how it goes in the long run

closed 2024-02-14 by supermerill