supermerill/SuperSlicer #4078: Start GCode twice

sboehle, 2024-01-18:
### What happened?

With the latest nightly build (more test & fixes to fill flow. #667) under win 11 the start gcode is added twice in the beginning of the gcode file:

; object:{"name":"Voron_Design_Cube_v8","id":"Voron_Design_Cube_v8.STL id:0 copy 0","object_center":[185.543878,153.209080,0.000000],"boundingbox_center":[185.543878,153.209080,15.000000],"boundingbox_size":[42.426407,42.426407,30.000000]}
EXCLUDE_OBJECT_DEFINE NAME=Voron_Design_Cube_v8_STL_id_0_copy_0 CENTER=185.543878,153.209080 POLYGON=[[164.33,153.21],[168.22,149.32],[181.30,136.24],[185.54,132.00],[189.79,136.24],[206.76,153.21],[203.89,156.08],[189.79,170.18],[185.54,174.42],[183.94,172.82],[182.09,170.97],[168.57,157.45],[164.33,153.21]]
; plater:{"center":[185.543878,153.209080,0.000000],"boundingbox_center":[185.543878,153.209080,15.000000],"boundingbox_size":[42.426407,42.426407,30.000000]}

M73 P0 R50
M73 C5
; custom gcode: start_gcode
SET_PRINT_STATS_INFO TOTAL_LAYER=150
print_start EXTRUDER=250 BED=110 PRINT_MIN=163.756,131.422 PRINT_MAX=207.331,174.997 FILAMENT=ABS
; custom gcode end: start_gcode
;TYPE:Custom
; custom gcode: start_gcode
SET_PRINT_STATS_INFO TOTAL_LAYER=150
print_start EXTRUDER=250 BED=110 PRINT_MIN=163.756,131.422 PRINT_MAX=207.331,174.997 FILAMENT=ABS
; custom gcode end: start_gcode

G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
ACTIVATE_EXTRUDER EXTRUDER=extruder
; custom gcode: start_filament_gcode
; Filament gcode
; Macro syntax: https://help.prusa3d.com/article/macros_1775

### Project file & How to reproduce

simply slice anything and in the gcode file, start gcode will be twice

### Version

latest nightly build

### Operating system

win 11

### Printer model

Voron 2.4

supermerill, 2024-01-18:
I'll launch a nightly compilation right now.

sboehle, 2024-01-18:
The latest nightly build fixed the issue. Thank you very much

closed 2024-01-18 by sboehle