supermerill/SuperSlicer #4079: Custom GCODE from Printer settings not honored

dpeart, 2024-01-18:
### What happened?

In my printer custom Gcode I have:
`START_PRINT EXTRUDER_TEMP={first_layer_temperature[initial_tool]} BED_TEMP=[first_layer_bed_temperature] MATERIAL=[filament_type] SIZE={first_layer_print_min[0]}_{first_layer_print_min[1]}_{first_layer_print_max[0]}_{first_layer_print_max[1]} INITIAL_TOOL={initial_extruder} LAYERS={total_layer_count} NOZZLE_SIZE={nozzle_diameter[0]}

STOP_LED_EFFECTS LEDS="neopixel:caselight"

SET_LED_EFFECT EFFECT=busy
SET_LED_EFFECT EFFECT=progress_bar

SET_PRINT_STATS_INFO total_layer={total_layer_count}`

I use START_PRINT instead of PRINT_START.

When I get the gcode it looks like:
`;TYPE:Custom
; custom gcode: start_gcode
PRINT_START

;; If you are passing variables to PRINT_START, comment out above line and uncomment the last 3 lines instead.
;; Guide: https://github.com/AndrewEllis93/Print-Tuning-Guide/blob/main/articles/passing_slicer_variables.md
;; Make sure the variable names match (for example if you use "EXTRUDER" instead of "HOTEND")
`

When I run the code I get the error that PRINT_START unknown as in my setup it is START_PRINT.  

This worked previously.

### Project file & How to reproduce

[[c]_Coupler_Block_xN.zip](https://github.com/supermerill/SuperSlicer/files/13977442/c._Coupler_Block_xN.zip)

### Version

2.5.59.6

### Operating system

windows 11

### Printer model

voron 2.4 with afterburner

supermerill, 2024-01-18:
when loading your project, i have that:
[image]

I don't really understand what you did and what you want.

dpeart, 2024-01-18:
OOPS, I was trying different things.  Let me re-upload

dpeart, 2024-01-18:
Well I think that is part of the problem.  Here is what my custom gcode looks like in the tool when I save the 3mf

[image]

But when I open the saved 3mf I get the default custom gcode setting like you show.

dpeart, 2024-01-18:
Here is a copy of the text:
`START_PRINT EXTRUDER_TEMP={first_layer_temperature[initial_tool]} BED_TEMP=[first_layer_bed_temperature] MATERIAL=[filament_type] SIZE={first_layer_print_min[0]}_{first_layer_print_min[1]}_{first_layer_print_max[0]}_{first_layer_print_max[1]} INITIAL_TOOL={initial_extruder} LAYERS={total_layer_count} NOZZLE_SIZE={nozzle_diameter[0]}

STOP_LED_EFFECTS LEDS="neopixel:caselight"

SET_LED_EFFECT EFFECT=busy
SET_LED_EFFECT EFFECT=progress_bar

SET_PRINT_STATS_INFO total_layer={total_layer_count}`

supermerill, 2024-01-18:
Can't reproduce, even when starting from a voron printer.
So, i need to be in the same state as you before you save.
Can you zip your configuration directory (you get to it via help -> show configuration directory) and drop it here?
And then the minimum manual steps to fail to save your custom gcode?

that way, i can really try to replicate your issue as much as possible.

dpeart, 2024-01-18:
[SuperSlicer.zip](https://github.com/supermerill/SuperSlicer/files/13978699/SuperSlicer.zip)

I then load any STL, slice and export gcode.

[image]

dpeart, 2024-01-18:
I don't know if this helps, but I also get a skirt, even though it is disabled.

dpeart, 2024-01-24:
Thanks for looking into it.
I recreated all my profiles from scratch and things have been working well since then.  This matches what you suspect.

closed 2024-01-24 by dpeart

supermerill, 2024-01-25:
I looked a bit why it's not possible to edit.
The problem is PresetBundle.cpp:l655 `out.apply(this->filaments.get_edited_preset().config);`
the filament preset has entries for everything, and so will overwrite everything.
exactly what you said, neophyl.
There already is some protections against that kind of problem, but there seems to be a way to slip through

supermerill, 2024-01-25:
Found the bug.
In preset.cpp:l1071:
`std::string incorrect_keys = Preset::remove_invalid_keys(config, default_preset.config);`
should be
`std::string incorrect_keys = Preset::remove_invalid_keys(preset.config, default_preset.config);`

Also submitted a pr to prusaslicer.