supermerill/SuperSlicer #4080: Fusion 360 Files unable to print

schwirzi, 2024-01-18:
### What happened?

If i tranfer a part from Fusion360 directly to Superslicer, slice it and send it to my voron, my printer goes in to Pause mode and do nothing. This happens only if i use SS Version 2.5.59.3 or 2.5.59.6. with F360. If i use 2.4.58.5 i can print my parts normaly.
A normal stl file from somewhere is printable with the 2.5 versions. 

### Project file & How to reproduce

[2.4 and 2.5 Parts.zip](https://github.com/supermerill/SuperSlicer/files/13979870/2.4.and.2.5.Parts.zip)

### Version

2.5.59.6

### Operating system

Windows 10

### Printer model

voron 2,4 with stealthburner

supermerill, 2024-01-18:
Can you send me your project file (the one you try with the 2.5.59.6)?
I think the problem is a unicode issue, like for #4070.
it may go away if you remove the 'Label objects' option.

schwirzi, 2024-01-19:
[Körper1.zip](https://github.com/supermerill/SuperSlicer/files/13989825/Korper1.zip)
there is the project file.
And your right. As i removed the Label objects option, i can print my file. For some check i enable it and its unprintable.

legend069, 2024-01-19:
> [Körper1.zip](https://github.com/supermerill/SuperSlicer/files/13989825/Korper1.zip) there is the project file. And your right. As i removed the Label objects option, i can print my file. For some check i enable it and its unprintable.

i would recommend raising an issue for this with klipper if it hasn't already been addressed.

ContainsNoMilk, 2024-01-21:
I think the problem is the "ö". I also can't print files with ä ö ü in the name.

supermerill, 2024-01-24:
just tested, bug in 2.5.59.6 and fixed in nightly.

supermerill, 2024-01-24:
I also encourage you to set your windows in utf-8 (beta) mode.

schwirzi, 2024-01-24:
Actually I changed my öäü letters and it's worked. German problems^^. And give the utf8 mode a try.
Thanks for your help.

closed 2024-02-14 by supermerill