supermerill/SuperSlicer #4081: [Feature request] Custom G-code section in Print Settings

Murrdo, 2024-01-18:
It would be very convenient to have a custom g-code section in the Print Settings section.
It would be possible to create print profiles for frequent tests/calibrations.

supermerill, 2024-01-18:
Add it to to print variable
then in your gcode, you can add it where you want.
in print custom variable (note tab)
```
start_gcode_print="; start gcode from print profile\nM42\nG42 S42\n; end start gcode from print profile"
```
where you want to put it in your printer custom gcode (like the start gcode)
```
{default_string(start_gcode_print,"")}{start_gcode_print}
```
if it's not present in the print setting, it will just write an empty line.
note: you need the latest nightly for this to work correctly.

Is it enough?
I think this way it's more versatile. It avoid the need a hundreds of specialized unused fields.

Murrdo, 2024-01-19:
Thank you, this is more than enough)

Murrdo, 2024-01-20:
I’ll leave here an example of how it works, I didn’t understand it right away.
1.
[image]
2.
[image]
Result
[image]

closed 2024-02-14 by supermerill