supermerill/SuperSlicer #4082: s_wall_thickness is not setting values as per description

veehexx, 2024-01-18:
### What happened?

I'm not sure if this is a problem with some overextrusion i'm seeing recently (not sure if slicer or print - still trying to figure it out!!) but i've noticed the following along with various nightly fixes with flow/extrusion related issues:

[image]

in the case of a 0.4mm nozzle (as defined with nozzle_diameter[0]) and as per the description of wall thickness, it states it must be between nozzle size and 2*nozzle size; so 0.4-0.8mm.

Trying some random values, it appears it also allows it substantially over 2x nozzle, but also seems to be doing something wrong with calculations as 1.299999 results in a 0.91 for 1 wall, or 1.78mm for 2 walls : 
[image]

### Project file & How to reproduce

set 0.4mm nozzle.

adjust wall thickness values inside description limits.
adjust wall thickness values outside of description limits

### Version

GTK3 ubu64 nightly 14441cf AppImage

legend069, 2024-01-19:
this is likely a floating point calculation error
can reproduce on 2.4.58.5 - windows
typing "1.3" into s_wall_thickness pressing enter and it auto corrects itself to 1.2999999x

veehexx, 2024-01-19:
it wasnt so much the rounding error, but the fact the description shows it can only be UPTO 2x nozzle (so 0.8 max wall thickness).

same 0.4mm nozzle with 1.1mm thickness = 1.14mm for 1 perimeter which isnt that far off 3x nozzle size:
[image]

supermerill, 2024-01-23:
There was an error.
The external perimeter half-spacing was used as a full spacing (so it was a bit bigger than expected).
fixed for next version.

closed 2024-02-14 by supermerill