supermerill/SuperSlicer #4083: Missing Walls/Lines on thin wall

blackfield3k, 2024-01-20:
### What happened?

After slicing with the Prusa Standart Settings (MK3S+, 0,15 Layer, Prusament PLA) two spots are missing walls after the first layer.

[image]
(One ear and the mouth)

I tried it with the latest stable and the newest pre release. 

Also changed Print settings, filament and printer (voron 2.4 and MK3S+, both with Klipper)

PrusaSlicer has no problem with it and the stl is also fine.
File was also reported as good by others.

### Project file & How to reproduce

[Gengar.zip](https://github.com/supermerill/SuperSlicer/files/13998902/Gengar.zip)

Used the usual workflow:
Load file
Select profile
Slice

No fancy slice settings or modifications

### Version

2.4.58.5 and 2.5.59

### Operating system

Windows 10

### Printer model

MK3S+

neophyl, 2024-01-21:
Prusa Slicer uses Arachne by default so it will pick up and slice thinner wall sections.  If you set PS to Classic slicing mode I suspect you will have the same missing sections.

SuSi does not use Arachne by default, even on versions that have Arachne.  When you select arachne then it slices ok for me with no missing walls.

The same can be done with classic slicing if you turn on thin walls.  

Projects/profiles can not be swapped 1:1 between SuSi and PS.  While most of the settings are common you will find a few that are not and that even have opposite effects.  You should carefully check ALL settings when doing so.

blackfield3k, 2024-01-21:
That did the trick! 

Unfortunaly 2.4 has no Arachne support and 2.5.59.6 crashes randomly. I think i know why but cant reproduce it.

Turning on thin walls also works but you have to play with the line with on this specific stl

Thanks for your help and this can be closed.

closed 2024-01-21 by blackfield3k