supermerill/SuperSlicer #4089: Nozzle crash into bed After slicing

Nesquirt, 2024-01-25:
I found a strange bug/corruption in stl, i've only changed infill and perimeters,when i started printing, the start gcode was executed well, but when it's the time of the 3d model, nozzle crash into bed..in a very strange way.
I've Bltouch and all is fine.
So i tryed to do mesh bed leveling, and was ok, so i think was the model and i switched to s3d and cura to check if model is ok, on this two slicer the object it's printed as expected, the nozzle doesnt hit the bed.
I've tryied to do a re-slice with SS but nothing, same problem..start printing, start gcode it's ok, when it's the time of the object the nozzle hit the bed.
Microstep and mesh bed leveling are fine...it's the first time it happen to me, i also check all belt, screw, guide, and all its okay, i think there is somenthing into the slicing that is corrupted.
adding images of my settings can help maybe
There is also the config file and gcode screenshot and machine limits.
Machine limits are the same on the printer

!!>> Project File <<!!
[PIS_edit.zip](https://github.com/supermerill/SuperSlicer/files/14057053/PIS_edit.zip)
[SuperSlicer_config_bundle.zip](https://github.com/supermerill/SuperSlicer/files/14057115/SuperSlicer_config_bundle.zip)

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version: 2.5.59
 - Printer model: SelfBuilt BLV AM8
 
 
 
 
[image]
[image]

supermerill, 2024-01-26:
can't see anything wrong.
Can you send me the good gcode file from cura?

Nesquirt, 2024-01-27:
[Corpo3.zip](https://github.com/supermerill/SuperSlicer/files/14072557/Corpo3.zip)
Now i dont have cura with me, only the s3d one, ill upload cura in some hours

supermerill, 2024-01-27:
Found
your start gcode doesn't set `G90`, so maybe your firmware think your gcode is in relative mode, and so when the printer pass in absolute mode, then it think it's much higher, and so crash the nozzle when it goes to printing height.

just put `G90` in the first line of your custom gcode.
It's also a good idea to set `M82` (absolute extrusion, as i think it's what you wrote) and `G21` (millimeters units)

Nesquirt, 2024-01-27:
Ill give a try,but i have **use relative E distance** unchecked on SS,and also on other slicer i have unchecked,in this way it doesnt told in gcode that is not in relative mode?

supermerill, 2024-01-27:
It is, but it's set after your custom start gcode if you don't set it yourself.
Some printer need very specific start gcode before that to be initialized.

Nesquirt, 2024-01-29:
For now work well,ill update issue if is possible,after more testing

Nesquirt, 2024-01-29:
@supermerill if i found maybe a bug in slicing i should open a new issue? Or can i contact you in dm? As telegram?

supermerill, 2024-01-30:
you can open a new issue.
If you really need to discuss, you can join the [discord](https://github.com/supermerill/SuperSlicer/issues/611#issuecomment-907833287) channel.

Nesquirt, 2024-02-02:
[Reverse_bottom_bowden_2020.zip](https://github.com/supermerill/SuperSlicer/files/14132522/Reverse_bottom_bowden_2020.zip)
Got the same issue...i noticed that on small part this issue happen a lot, idk, this is the gcode, i also add the line you suggest to me....

i also upload the one from s3d that works well
[Simplify_Reverse_bottom_bowden_2020.zip](https://github.com/supermerill/SuperSlicer/files/14132545/Simplify_Reverse_bottom_bowden_2020.zip)

The issue happen after the two purge line when moving to position for start printing

Nesquirt, 2024-02-06:
@supermerill no help?🤔

supermerill, 2024-02-07:
I don't understand why your printer crash your nozzle.

only guess is the 
```
M117 Printing!
M107
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
```
section resetting the z height.
You can try to add it in your s3d gcode, just before `; feature skirt`, to see if it broke again.
To prevent it to be printed in susi, check the "only custom start gcode" in the custom gcode tab on your printer configuration.

Nesquirt, 2024-02-28:
https://youtube.com/shorts/Ds2AScCrxjA

@supermerill today after some 2 weeks of rebuilding 3d printing, got the same issue, i add a video.
i switched also to klipper, i dont know why this happen but is very frustrating

[Test Cube.zip](https://github.com/supermerill/SuperSlicer/files/14427407/Test.Cube.zip)
[SuperSlicer_config_bundle.zip](https://github.com/supermerill/SuperSlicer/files/14427414/SuperSlicer_config_bundle.zip)

supermerill, 2024-02-28:
Are you sure it's "crashing" ? it's maybe a bit too close (is the z-height really hight enough?), or the bed is maybe not even enough.
You can use the printer's "min height for travel" (`lift_min`) and put it at 1mm. That way it won't scratch the surface when moving to the objects.

Nesquirt, 2024-02-28:
@supermerill using the settings on slicer for now seems to works well,maybe when it goes from purge line to model to print the z raising is slow,doing what you do seems to fix,but i've a question.. it raise only if z is lower than 1 mm or is like z hop and raise always also on retraction?

supermerill, 2024-02-29:
it's a minimum, so it won't enfroce any higher z. It also only works for going to& from an object. so if you have only one part, it's only used to go to & from the skirt.

Nesquirt, 2024-03-01:
thanks mate for now works well!