supermerill/SuperSlicer #4094: Pressure Advance Settings with Ramming

Mawe066, 2024-01-29:
**Is your feature request related to a problem? Please describe.**
When Ramming with Multiple Colors is activated, it sets Extruder Pressure Advance to 0 without Extruder information, which causes an "unable to infer stepper" error in Klipper.

**Describe the solution you'd like**
Add Extruder information to pressure Advance setting for CP Tool head Unload

Maybe I am just to dump or overlooking something, but I had to disable ramming because it's triggering an error in Klipper.

By the way, is there any time zone you can estimate version 2.7? I would love to have organic supports.

supermerill, 2024-01-29:
> By the way, is there any time zone you can estimate version 2.7? I would love to have organic supports.

I think end of February is not too pessimistic.
My original plan was to finish the 2.7 by december, to use that in january.
But I couldn't make it, so I had to dedicate my january to stabilize the 2.5.
I think the 2.5 is now in a good shape, and i can return to 2.7 merge.

Mawe066, 2024-01-29:
Sounds awesome, definitely looking with excitement to it and that you started development again after almost a year of silence around Super slicer.
Tuning my wipe tower for full 3 days and still can't figure it out yet. Sometimes my filament gets stuck, maybe because of bad tip shaping, which is weird, because I use an cht nozzle. There is not really a Tip 😅

Mawe066, 2024-01-30:
Quick update:
I used the wrong expressions. In the filament gcode I used {tool_name} instead of [tool_name].
Now it works as expected

closed 2024-01-30 by Mawe066

supermerill, 2024-01-30:
It's weird that it's different. `[var_name]` is just the old syntax for `{var_name[current_extruder]}`
and `{var_name}` also is translated to `{var_name[current_extruder]}`

Mawe066, 2024-01-30:
I don't know, but with EXTRUDER={tool_name} it does not work and with EXTRUDER=[tool_name] it works perfectly fine.
Here a picture of my setup.
[image]

Mawe066, 2024-01-31:
Sorry for my late response, i dont had the time since.
The Issue that triggerd an "unable to infer stepper motor" was caused by SET_PRESSURE_ADVANCE ADVANCE=0.0, which is caused by:

> ;Filament GCODE
;PLA PA
{if tool_name == "extruder"}
SET_PRESSURE_ADVANCE ADVANCE=0.750 SMOOTH_TIME=0.040 EXTRUDER={tool_name}
{endif}

>{if tool_name == "extra_extruder1"}
SET_PRESSURE_ADVANCE ADVANCE=0.750 SMOOTH_TIME=0.040 EXTRUDER={tool_name}
{endif}

>{if tool_name == "extra_extruder2"}
SET_PRESSURE_ADVANCE ADVANCE=0.750 SMOOTH_TIME=0.040 EXTRUDER={tool_name}
{endif}

>{if tool_name == "extra_extruder3"}
SET_PRESSURE_ADVANCE ADVANCE=0.750 SMOOTH_TIME=0.040 EXTRUDER={tool_name}
{endif}

I changed it to following, what for me, fixed my issue:

> ;Filament GCODE
;PLA PA
{if tool_name == "extruder"}
SET_PRESSURE_ADVANCE ADVANCE=0.750 SMOOTH_TIME=0.040 EXTRUDER=[tool_name]
{endif}

>{if tool_name == "extra_extruder1"}
SET_PRESSURE_ADVANCE ADVANCE=0.750 SMOOTH_TIME=0.040 EXTRUDER=[tool_name]
{endif}

>{if tool_name == "extra_extruder2"}
SET_PRESSURE_ADVANCE ADVANCE=0.750 SMOOTH_TIME=0.040 EXTRUDER=[tool_name]
{endif}

>{if tool_name == "extra_extruder3"}
SET_PRESSURE_ADVANCE ADVANCE=0.750 SMOOTH_TIME=0.040 EXTRUDER=[tool_name]
{endif}

I dont knof if thats on purpose, or its in fact a bug, but there you go

supermerill, 2024-02-01:
can you send me the generated gcode by both versions?

Mawe066, 2024-02-01:
Hmm thats weird... I just inspected the Gcode and there is no difference, but one works, one not... I dont get it..

supermerill, 2024-02-01:
they are really similar? maybe you inserted an invisible character.

Mawe066, 2024-02-01:
What do you mean with invisible character?

supermerill, 2024-02-01:
an example: https://emptycharacter.com/