supermerill/SuperSlicer #4095: Unable to open SuperSlicer on MacOS now, can't even open old versions

L1S9D7M3T, 2024-01-29:
### What happened?

every time i open SuperSlicer, i get an error that i cannot close that states:

Failed to load image from file "/Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer".

with the following 2 errors with red circles next to them:

- This is not a PNG file.
- Failed to load image from file "/Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer".

when i open the actual executable file in Terminal, it gives the following errors:

Last login: Mon Jan 29 13:21:37 on ttys001
LSDMT@MBP18 ~ % /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer ; exit;
[2024-01-29 13:21:38.868469] [0x00007ff8578b8b80] [trace]   Initializing StaticPrintConfigs
2024-01-29 13:21:38.962 SuperSlicer[8899:144543] WARNING: Secure coding is not enabled for restorable state! Enable secure coding by implementing NSApplicationDelegate.applicationSupportsSecureRestorableState: and returning YES.

[2024-01-29 13:21:39.555273] [0x00007ff8578b8b80] [error]   3dx drivers module loading error: dlopen(/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient, 0x0005): tried: '/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient' (no such file), '/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient' (no such file), '/System/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient' (no such file, not in dyld cache)

[2024-01-29 13:21:39.651814] [0x00007ff8578b8b80] [error]   Unable to load bed texture: C:\Users\robth\AppData\Roaming\SuperSlicer\Troodon Bed Texture.png

[2024-01-29 13:21:39.882737] [0x00007ff8578b8b80] [error]   stl_open_count_facets: The file /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer has the wrong size.

[2024-01-29 13:21:40.525815] [0x00007ff8578b8b80] [error]   stl_open_count_facets: The file /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer has the wrong size.

[2024-01-29 13:21:40.532385] [0x00007ff8578b8b80] [error]   stl_open_count_facets: The file /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer has the wrong size.

[2024-01-29 13:21:40.648558] [0x00007ff8578b8b80] [error]   stl_open_count_facets: The file /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer has the wrong size.

[image]
[image]

### Project file & How to reproduce

can't - because i can't even open SuperSlicer

### Version

2.5.59.6 nightly - i tried them all and even 2.5.59.4

### Operating system

macOS 14.2.1 

### Printer model

Troodon Core XY 400

supermerill, 2024-01-31:
can you try with this one?
https://github.com/supermerill/SuperSlicer/actions/runs/7723088019/artifacts/1208304988
I fixed some issue in a file used by macos to get some icons. Maybe that's that.

L1S9D7M3T, 2024-01-31:
Merill,

it works now - looks like i am back in business! thank you sir!

i got these messages when i loaded it, so there must be a bunch of new settings?

￼

- Ben

L1S9D7M3T, 2024-01-31:
Merill,

it works now - looks like i am back in business! thank you sir!

i got these messages when i loaded it, so there must be a bunch of new settings?

[image]

supermerill, 2024-01-31:
just a nightly with a big rework that wasn't has enough of use to be iron out yet, see #4098.
Will be solved this night or the coming ones.

L1S9D7M3T, 2024-01-31:
ok - THANK YOU for the prompt response and all your hard work!

L1S9D7M3T, 2024-02-05:
the error came back and I've been unable to open SuperSlicer since (i think) last Friday. i even downloaded the latest nightly MacOS build but the 'Failed to load image' issue persists. i re-downloaded the version from the link you posted above and the error is still there. is it somehow a roaming error?

L1S9D7M3T, 2024-02-09:
just downloaded the 2.5.59.7 version - still getting the same error:
[image]
[image]

supermerill, 2024-02-10:
I'll make you a build with logs to pinpoint the issue.

L1S9D7M3T, 2024-02-15:
still having the same issue on the latest build (2.5.59.8). same error as above. if i open the SS executable file with terminal, you can still see these errors:
[image]

comatose-tortoise, 2024-02-17:
Any way to get around this?

mattolenik, 2024-02-18:
Just use the Intel build with Rosetta. Works great on my M1.

comatose-tortoise, 2024-02-18:
I'm using the intel build on my intel mac.. Not working.

mattolenik, 2024-02-18:
> I'm using the intel build on my intel mac.. Not working.

Do you have Rosetta installed? Normally macOS would prompt you to install Rosetta when you launch an x86 app, but maybe these builds being unsigned is preventing that from working.

Try this in your terminal:

`softwareupdate --install-rosetta`

comatose-tortoise, 2024-02-18:
Isn't Rosetta for running Intel stuff on Apple Silicon?

mattolenik, 2024-02-18:
> Isn't Rosetta for running Intel stuff on Apple Silicon?

Oh I'm sorry, I misread what you wrote. 🤦🏻 You're right, that won't help you there.

L1S9D7M3T, 2024-02-18:
> Just use the Intel build with Rosetta. Works great on my M1. I just installed the latest release on Sonoma 14.3.1.

using an emulator does not solve the issue

supermerill, 2024-02-21:
hi, can you try the build https://github.com/supermerill/SuperSlicer/actions/runs/7973682768 ?
it doesn't resolve the issue but has more debug output.

slashbeast, 2024-02-23:
> hi, can you try the build https://github.com/supermerill/SuperSlicer/actions/runs/7973682768 ? it doesn't resolve the issue but has more debug output.

The linked build start up just fine on ARM64 MacOS, and basic slicing seems to work as well, no errors or crashes whatsoever. Tested on MacOS 14.3.1 with ARM64 M3 Pro SoC.

comatose-tortoise, 2024-02-23:
> hi, can you try the build https://github.com/supermerill/SuperSlicer/actions/runs/7973682768 ? it doesn't resolve the issue but has more debug output.

This build starts and slices fine on my 2019 Intel Mac. No PNG file errors to be seen.

L1S9D7M3T, 2024-02-25:
haven’t been able to try yet as i am out of town on business. will let you know this upcoming week.

L1S9D7M3T, 2024-02-27:
i saved the config bundle + printers on my PC and deleted everything 'superslicer' that i could find on my mac and installed the 2.5.59.9 from the link above. seems to be working for now. maybe a cache or preference file got screwed up somewhere. i will be testing it out this week and will reply back if any issues are found.

thanks, merril and everyone else who worked on it for us.

SirOink, 2024-03-02:
I figured what is causing the problem. All versions work for me now, just checked. 
It could be a permissions problem or a chmod one. I believe superslicer needs to request permission to access the users folder; macos/macs are made to be retard-proof, so non-signed apps require heavy borders. Still love 'em.

for macos users there's a temp fix... go to
`/Users/{USERNAME}/Library/Application Support/SuperSlicer/Printer`

and in each .ini file search for the following lines
`bed_custom_model = {PATH/File.stl}`
`bed_custom_texture = {PATH/File.png}`

and delete the url's to your custom texture files, or delete then copy/paste:
`bed_custom_model =`
`bed_custom_texture =`

To replicate the problem, put the texture model's in the macos documents or downloads folder, change the above stated paths to it and restart superslicer. Will require a force quit due to erroring-out.

I haven't tried putting the texture files in a different folder yet (preferably one SS has full permission to read, like Application Support/Superslicer) but I surmise that may work as a patch until meril has the time to fix it permanently

Dissfall, 2024-05-23:
A script that does what @SirOink described above:

```bash
cd /Users/$USERNAME/Library/Application\ Support/SuperSlicer/Printer; sed -i -e "s/bed_custom_model.*/bed_custom_model \=/g" *.ini; sed -i -e "s/bed_custom_texture.*/bed_custom_texture \=/g" *.ini;
```

Just put it in terminal and hit enter.

viniavila, 2024-05-28:
Based on @SirOink solution, I could solve this issue by direct editing my printer's ini file AND BEING ABLE TO LOAD THE TEXTURE AND STL MODEL. That's the procedure:

1. Open in Terminal the INI file (I used vim, but you can choose any text editor of your choice):
`vim /Users/$USERNAME/Library/Application\ Support/SuperSlicer/printer/<my_custom_printer_ini_file>.ini`

2. Search the lines containing `bed_custom_model = ` and `bed_custom_texture = `

3. After `=` put the FULL PATH to your desired files (stl for the model, png for the texture)

4. Save the file and exit. Try to reload SuperSlicer.

I haven't tested the path with variables like `$USERNAME` or using the tilde (`~`) to replace `/Users/$USERNAME`. You can be free to test them. I printed the result below.
[image]