supermerill/SuperSlicer #4100: [FR] internal perimeter variable speed change

legend069, 2024-02-01:
This suggestion came from a friend of mine.
There's a quality/dimensional accuracy issue that can occur when external /internal perimeters are printed with large difference in extrusion speeds.
Ie; internal 500mms external 100mms

Add a variable to dynamically change the internal perimeter speed * perimeter count.

Example;
Perimeter 0 100mms
Perimeter 1 120mms
Perimeter 2 140mms
Repeat increasing internal perimeter speed till it reaches total perimeter count/ target internal perimeter speed

supermerill, 2024-02-01:
A "max speed change for perimeter"

An exemple:
 . ext. peri. speed to 50
 . int. peri. speed to 500
 . max peri speed change to 25
 . perimeter count to 3

then the second perimeter will have a speed of 75 and the third one a speed of 100

Is it what you want?

legend069, 2024-02-01:
> An example: 
ext. peri. speed to 50
int. peri. speed to 500
max peri speed change to 25
perimeter count to 3

> then the second perimeter will have a speed of 75 and the third one a speed of 100

yeah pretty much.
the internal perimeters could have a 'factor' with it to raise each internal peri by x speed

supermerill, 2024-02-01:
> the internal perimeters could have a 'factor' with it to raise each internal peri by x speed

do you have an exemple?

Mike1618, 2024-02-12:
Yeah, so the intention here is being able to have more granular control over perimeter speeds. At the moment internal perimeters are only defined as a perimeter that is not the external. 

The issue, as it currently stands is: for example, if I print 4 walls, settings for external is 100mm/s and internal 500mm/s. In this case, we are setting the external wall slower(100mm/s) to print at a better quality, while the internal faster (500mm/s) to save print time. The issue is, that defects(such as resonance, blobs, bubbles, etc) from one wall can affect an adjacent wall. Walls that are furthest away from the external perimeters(or closest to the infill) have the least effect on the external wall, but printing all 3 internal walls at the same high speed (500mm/s) can translate/mirror/echo defects all the way out to the outermost wall, the external perimeter, if printing the wall order from inner to outer. Even though the outermost perimeter is printed at a slower 100mm/s it will still translate/echo any defects of the adjacent wall next to it.

If we had a way to stagger/ or step down wall speeds more gradually,  it would allow the printer to smooth out defects before the last outer wall is printed. 

I think we should have a function to control 5 levels of speed of the outermost perimeters, to be activated when required. 
So for example, going from the outermost wall inwards:
External perimeter speed( or outermost wall)-> 
1st internal perimeter speed( which is adjacent, next to, the external perimeter) -> 
2nd internal perimeter speed-> 
3rd internal perimeter Speed-> 
other internal perimeter Speed (which is 5th and all other walls going towards the infill).

So with this, printing 4 walls, i can have the:
External perim:100mm/s
1st interna periml: 200mm/s
2nd internal perim: 300mm/s
3rd internal perim: 500mm/s

This would be useful also for very thick walls, say for example i wanted to print a brick with 10mm thick walls at 0.5mm layer widths, whic is20 walls. I could set 
External perim:100mm/s
1st internal: 200mm/s
2nd internal: 300mm/s
3rd internal: 400mm/s
Other internal walls(which is the 5th to 20th wall from the external): 500mm/s.

With thick walls (high perimeter count) like this, the walls that are adjacent to the outer perimeter are more like external perimeters but currently are defined as internal walls with only a one speed setting to control all perimeters up to the external perimeter, which means, in this example i would have to do all 19 internal perimeters at one speed(500mm/s), before drastically dropping to 100mm/s for the external outermost wall. 

The solution mentioned above of 
_“A "max speed change for perimeter"
An exemple:
 . ext. peri. speed to 50
 . int. peri. speed to 500
 . max peri speed change to 25
 . perimeter count to 3
then the second perimeter will have a speed of 75 and the third one a speed of 100”_
This can work, but there might be an issue where one of the calculated speeds falls on a speed at which the motors resonate at.

So with the above solution using the “max speed change perimeter”
. ext. peri. speed to 100
 . int. peri. speed to 500
 . max peri speed change to 100
 . perimeter count to 4

Which would give me:
External perim:100mm/s
1st interna periml: 200mm/s
2nd internal perim: 300mm/s
3rd internal perim: 500mm/s

What if, I want the 1st internal perimeter to not fall exactly on 200mm/s, as it resonates at that speed, and want a speed of 180mm/s or 220mm/s instead? So a “max peri speed change” makes it hard to accomplish this, rather than being able to set absolute values for each wall.