supermerill/SuperSlicer #4104: Support Interface missing on some overhangs

78MC-Prime, 2024-02-03:
### What happened?

I have noticed that after build 2.5.59.3, support interface is missing for some overhangs.  Is there a setting I am missing in versions later than 2.5.59.3 that restores the support interface for all overhangs?
[image]
[image]

### Project file & How to reproduce

Here is the STL model used to discover this issue.

[Orbiter_V2_Mount-Orbiter_Head_Mount.zip](https://github.com/supermerill/SuperSlicer/files/14152065/Orbiter_V2_Mount-Orbiter_Head_Mount.zip)

### Version

2.5.59.3 to 6

### Operating system

Windows 10

### Printer model

Tronxy X5SA Pro

supermerill, 2024-02-07:
[image]
can't reproduce.
can I have your project file instead of just the stl, to be able to reproduce the issue?

78MC-Prime, 2024-02-07:
Supermerill,
That's interesting, maybe I have the wrong build.  I just reproduced the issue in the software.
[image]
[image]
[Orbiter_V2_Mount-Orbiter_Mount_Side.zip](https://github.com/supermerill/SuperSlicer/files/14195562/Orbiter_V2_Mount-Orbiter_Mount_Side.zip)

closed 2024-02-07 by 78MC-Prime

supermerill, 2024-02-07:
can reproduce with 2..5.59.6
can't reproduce on my nightly, so seems already fixed.
seems like it's time for a new release.

78MC-Prime, 2024-02-07:
Is the nightly build available for me to use?  I like the support interface on your post.  Looks like it would produce a cleaner surface.

supermerill, 2024-02-07:
in the action menu below, choose your os, then the download link is a bit below.

78MC-Prime, 2024-02-07:
Thank you for addressing this.