supermerill/SuperSlicer #4105: Exclude certain print speeds

RURon, 2024-02-03:
I've come to look at undesired resonances at certain speeds with my CoreXY printer (using Klippain macro) and now that I got rid of all unwanted motion speeds in the speeds settings in Superslicer, the only problem is when Superslicer reduces print speeds due to minimum layer time for a filament. Of course also relative print speeds may result in unwanted speeds.

So I would like to suggest an option to name speed intervals that should be avoided. As this would be printer (and not filament) specific, I suggest to include this into the printer settings (maybe machine limits?). One way could be a text input field where the user could enumerate unwanted speed intervals like "50-80,120-150,300-330" or anything more fancy of course.

While slicing, if print speed falls into one of those unwanted corridors, the next slower (or faster, maybe a user preference?) would be picked.

mara5a, 2024-07-04:
This is a good idea. But it will have to be a bit more sophisticated as the excluded speeds change depending on direction of the move for CoreXY printers I think.

RURon, 2024-07-04:
> This is a good idea. But it will have to be a bit more sophisticated as the excluded speeds change depending on direction of the move for CoreXY printers I think.

While I agree that there might be differences for each motion axis, from my printer (VCoreXY) I can tell that I can find speed bands where every motion is silent (which is probably a desired achievement to have a robust motion system that does not start to oscillate at some movements)

TheSlashEffect, 2024-07-04:
That is true, 45 degree angles are nastier than moving along the X or Y axes. However, taking this into account would make the feature very complicated for little gain.
Implementation would be difficult to implement/maintain, and bugprone afterwards unless plenty of things are refactored. In addition, users would have way too many parameters to tweak.
We don't know how each printer behaves to say 50 degree vs 45 degree movements. I believe simpler is better here.

That being said, I did have the idea of implementing a warning feature informing the user of fast moves near the 45 degree line.

supermerill, 2024-07-04:
If an extrusion has a speed between two ok band, the speed is reduce to the lower one, the nearest one?

mara5a, 2024-07-04:
> If an extrusion has a speed between two ok band, the speed is reduce to the lower one, the nearest one?

That could be user choice (lower, nearest, higher). But I'd guess default would be lower one

TheSlashEffect, 2024-07-12:
Good day, everyone.
I've a question to the more advanced printing people and the author (@RURon) since I started some experiments on this and cannot print at the moment:

How do the resonances look on arcs? while moving in an arc in 50mm/sec X and Y axis speeds are not constant, how does that look on the final print compared to the artifacts on a straight line?

RURon, 2024-07-13:
@TheSlashEffect I can only tell from real printing: I dont really see resonances when I tell Superslicer to print at a given speed (that was tested to be free of resonances), no matter what moves are carried out. You could probably come up with moves that would result in resonances, but I guess this feature request would rather try to eliminate most of resonant moves, not all of them. 
Maybe an approach could be to first try an easy implementation and in case there would be room left for improvement, that could be "escalated" later?

TheSlashEffect, 2024-07-13:
Thanks for the input. That is my plan, to work on this feature incrementally. Especially considering the minimum layer time code which interferes with this feature is complex and working alongside that would take some more time. I'm working on a simple version at the moment.

TheSlashEffect, 2024-07-18:
Ask, and you shall receive. A [beta version PR](https://github.com/supermerill/SuperSlicer/pull/4381) is up. In about a week I will resume work to make this work fully, alongside the minimum layer time. I've an open question there so all user's input on that along with anything else they would like changed is highly appreciated. Have a good rest of the day, everyone.