supermerill/SuperSlicer #4108: Modifiers such as speed and extrusion multiplier do not work

inovatorius, 2024-02-04:
### What happened?

Adding a modifier such as speed or extrusion multiplier to a part doesn't apply to it.

Settings:
[image]

Preview (speed):
[image]

Preview (extrusion multiplier):
[image]

### Project file & How to reproduce

[Flat sheet v3.zip](https://github.com/supermerill/SuperSlicer/files/14157305/Flat.sheet.v3.zip)

### Version

2.5.59.6

supermerill, 2024-02-07:
Fixed the print_extrusion_multiplier.
Can't see any issue with external perimeter speed.

inovatorius, 2024-02-08:
I saved the project and reopened SuperSlicer and now everything loads correctly. For whatever reason, it happens only when you start new project. I'll test the changes once they area ready, thanks a lot!

closed 2024-02-14 by supermerill