supermerill/SuperSlicer #4111: Layer time (log) become crazy

Nerobeus, 2024-02-08:
### What happened?

The Layer Time (log) is completly wrong on last version and the layer time goal still not good.

[image]

Here the previous version to compare 

[image]

### Project file & How to reproduce

[Part.zip](https://github.com/supermerill/SuperSlicer/files/14210696/Part.zip)

legend069, 2024-02-11:
maybe related to https://github.com/supermerill/SuperSlicer/issues/3890 ?

legend069, 2024-02-11:
your attached Zip file only contains the STL.
can you please attached the project file.

supermerill, 2024-02-11:
not fixed in current nightly, but in the next one. It needed a little more.

closed 2024-05-03 by supermerill