supermerill/SuperSlicer #4113: Partcooling fan to 100% i.s.o. to 30% setpoint (max_fan_speed) in case layer time < fan_below_layer_time

Arthur-de-Partuur, 2024-02-09:
### What happened?

The partcooling fan goes to 100% in stead of to the setpoint of 30% in case the layer time is below 10 seconds (fan_below_layer_time)

Setpoints:
[image]

Gcode preview:
[image]

In case I set the fan_below_layer_time to 0 the fan stays at the 25% (default fan speed) 
[image]

Gcode preview:
[image]

I would expect that the fan goes to 30% in stead of 100% as that is what is set at max_fan_speed

### Project file & How to reproduce

[Andrew Ellis EM-calibration cubes COARSE STEPS.zip](https://github.com/supermerill/SuperSlicer/files/14223649/Andrew.Ellis.EM-calibration.cubes.COARSE.STEPS.zip)

### Version

2.5.59.7

### Operating system

win11

### Printer model

Ratrig VCore 3.1-300

supermerill, 2024-02-10:
You have the fan kickstart activated. That bring the fan to &00% for a short amount of time.
But it's weird that you have a full layer of full fan speed.
I have that: 
[image]
I'll check if I modified something else that may create your issue.

Arthur-de-Partuur, 2024-02-11:
I am puzzled as well. With theoriginal .3mf file, I can reproduce the issue. If I open another project, the issue is not present.
[Andrew Ellis EM-calibration cubes MEDIUM STEPS 0860-1000.zip](https://github.com/supermerill/SuperSlicer/files/14232107/Andrew.Ellis.EM-calibration.cubes.MEDIUM.STEPS.0860-1000.zip)

supermerill, 2024-02-11:
I tried again with a fresh 59.7, in an empty configuration, and I still can't reproduce.

Arthur-de-Partuur, 2024-02-11:
I have loaded 2.5.59.7 with the default datadir (my regular profiles are on  non-default location, loaded with the --datadir switch)
I have opend the above .3fm and notice the problem is NOT there when fan_below_layer_time =5 or smaller. When set to  6 or higher the fan goes to 100% is.o. the max 30% max_fan_speed.

Paradisebaker, 2024-02-12:
I noticed something similar, not based on the fan_below_layer_time, but for fan increase for overhangs.
in my case, I want the fan to give 100% for overhangs and I have specified kickstart time.
As soon as the fan goes 100%, it doesnt go back down anymore.
Could it be that the kickstart time is longer than the duration of printing the overhang and this is kind of cancelling the fan reducing signal? with Klipper and high speed printing this might occur.
I also tried the 2.5.59.7, same thing.
I am not at home right now, but I can include the 3mf if wanted.

supermerill, 2024-02-12:
> I am not at home right now, but I can include the 3mf if wanted.

Yes.
And if somethign like that happens, can you try without any speedup/kickstart? If the output if good without, you can also include the two gcode files, as that's what I use anyway to create test for the fan mover.

Quarterblad, 2024-02-12:
Kickstart duration of 0.2s (0.2 is just my number) with checkbox  "only on overhangs"  is causing 100% fan speed on random block of layers, even if it`s layer doesn't have any overhangs. 
[image]
Same config and model sliced in 2.5.59
[image]

When I found that if "internal bridges" is set form "1", which was in all my old contigs, to "1" it will fix the problem.

Arthur-de-Partuur, 2024-02-12:
Changing the parameter fan_speedup_overhangs does not trigger any change with me.
fan_kickstart = 0 resolves the 100% fan-on. Setting it to 0.46shows the problem (so anything between 0 and 0.459 does not show it in my calibration-square example). Assessed Layertime for this layer is approx.10s.

However, after I restarted Superslicer due to a crash (when resizing the window it hangs,... probably a different issue :-)) the values are slighlty different:
Setting it to 0.459 shows the problem (so anything between 0 and 0.458 does not show it in my calibration-square example).

##YES Problem
[image]
[image]
[image]

##No Problem
[image]
[image]
[image]

Hope above helps,

supermerill, 2024-02-12:
Can I have the gcode files?

Arthur-de-Partuur, 2024-02-12:
[problem.zip](https://github.com/supermerill/SuperSlicer/files/14249366/problem.zip)

.3mf + gcode

On a different computer and I have problems to recreate (but I managed). That's why i included 3mf

supermerill, 2024-02-12:
I also need the gcode generated without the kickstart (so without the bug) to be able to compare them.

supermerill, 2024-02-12:
nevermind, I sliced with your project, and the two files are the same.
I guess the problem may come from the gcode viewer, and not th gcode.

supermerill, 2024-02-12:
I sliced your project, I can replicate the issue. thanks.

Arthur-de-Partuur, 2024-02-12:
I changed Slowdown_below_layer_time to 0
[No problem.zip](https://github.com/supermerill/SuperSlicer/files/14249569/No.problem.zip)

closed 2024-05-03 by supermerill