supermerill/SuperSlicer #4121: gcode preview layer 0 lets you scroll through the generated gcode

legend069, 2024-02-12:
### What happened?

having the gcode preview height set to 0, you can still scroll forward/back on the hidden scroll bar.
this could confuse some users thinking it's duplicating the first layer.

not present in 2.4.58.5 only later versions -2.5.59.7

going to layer 0 the gcode preview should start the scroll from the top of the file.
or should the preview only start from layer 1 ?

[image]

### Project file & How to reproduce

scroll down to layer 0 use arrow keys on keyboard left/right once file is sliced

[ss_test_body_project.zip](https://github.com/supermerill/SuperSlicer/files/14235874/ss_test_body_project.zip)

### Version

2.5.59.7

### Operating system

win10

legend069, 2024-02-12:
> That's funny. What do you do to scroll? I can't find a way to do it.

just the left/right arrow keys on the keyboard.
up/down arrow keys goes up/down in layers

supermerill, 2024-02-12:
I try to make the layer "0" what's before the start of the print (mostly what's inside the custom start gcode).

legend069, 2025-04-09:
in 2.7.61.0
with travel movements enabled this bug is still possible