supermerill/SuperSlicer #4122: Reporting of different printer parameters

macchinette, 2024-02-12:
### What happened?

As you can see from the video, moving from one profile to another indicates that there are different values, but moving to a different tab of that profile everything returns to normal without any differences.

This only happens for printer profiles.

### Project file & How to reproduce

https://github.com/supermerill/SuperSlicer/assets/79080486/a5f2e6f5-8210-4591-84a1-5bde36e5255b

[3DBenchy.3mf.zip](https://github.com/supermerill/SuperSlicer/files/14241562/3DBenchy.3mf.zip)

### Version

Version: 2.5.59.7 Build: SuperSlicer_2.5.59.7_2024-02-11

### Operating system

windows 10

supermerill, 2024-02-12:
Can I have your configuration directory? As this is about switching between presets.
You can create a new one and populate it so it exhibit the problem, if needed.

macchinette, 2024-02-12:
Tomorrow I will attach the complete bundle to you.

Bedeschi Cristian

> Il giorno 12 feb 2024, alle ore 19:01, Merill ***@***.***> ha scritto:
> 
> Can I have your configuration directory? As this is about switching between presets.
> You can create a new one and populate it so it exhibit the problem, if needed.

macchinette, 2024-02-13:
> Can I have your configuration directory? As this is about switching between presets. You can create a new one and populate it so it exhibit the problem, if needed.

[SuperSlicer_config_bundle 2_6.ini.zip](https://github.com/supermerill/SuperSlicer/files/14269706/SuperSlicer_config_bundle.2_6.ini.zip)