supermerill/SuperSlicer #4126: Additional Raft Parameters for Improved Large Volume Printers Reliability

Erabior, 2024-02-13:
**Is your feature request related to a problem? Please describe.**
I am coming from Cura due to a firmware upgrade on my printer at work. It is a large volume machine (450mmx450mmx565mm). I found that the way rafts are generated on superslicer are not as reliable as the rafts that I was able to create with the level of control given in Cura. Admittedly, the rafts I set up in Cura were inspired by those generated in Simplified3D. In essence, the first layer of the raft would be .6mm layer height (0.6mm nozzle btw), line width of ~1.0mm, line spacing of 1.8mm, and a speed of anywhere between 30mm/s and 60mm/s (speeds depending on the size of the part) the middle layers (typically 2 layers) would be 0.45mm thick, 0.6mm wide and 0.8mm apart, speed between 30-60mm/s, finally 1 top layer at .35 height, 0.6 width, 0.6 spacing (solid, no gap), speed of 45-60mm/s. Furthermore, the option to have the Wipe Tower printed the raft with the model would greatly improve the printing experience on large volume IDEX machines. The issue with increasing the wipe tower brim is an issue of Z-offset accuracy across the entire surface of the build plate. this can lead to a brim that is perfect, 'snowplowed', or too high off the bed leading to adhesion issues.

**Describe the solution you'd like**
I believe these settings, if possible, would greatly improve bed adhesion on large volume 3D printers:

Raft Top Layer Quantity
Raft Top Layer Height
Raft Top Line Width
Raft Top Line Spacing
Raft Top Speed

Raft Middle Layer Quantity
Raft Middle Layer Height
Raft Middle Line Width
Raft Middle Line Spacing
Raft Middle Speed

Raft Base Layer Quantity: determines the number of base layers to print.
Raft Base Layer Height: Sets the layer height for this section of the raft.
Raft Base Line Width: determines the width of all lines printed on this part of the base.
Raft Base Line Spacing: line spacing distance, I.E. if the line spacing is set to 0.8mm, the center of each line will be separated by 0.8mm. If the line width is 0.4mm, then there will be a 0.4mm gap between lines.
Raft Base Speed: print speed on the base layer of the raft.

Additionally, these changes under Wipe Tower category:

Add:
Wipe Tower Adhesion Type- Drop down: Brim/Raft Slaved to model adhesion type if "Merge with Model Adhesion" selected.
Merge with Model Adhesion- check box, when selected, generates brim/raft as if wipe tower was part of a merged model.

Change:
"Wipe tower brim width" to "Wipe tower adhesion width"

**Describe how it would work**
These settings would only be visible in "Expert" mode with the exception of the additional wipe tower settings, which I feel are both beneficial enough and straight forward enough for novice 3D printer operators. The raft settings would allow for a significantly higher degree of control over raft generation. Especially in my very specific use case where I almost need the first layer raft lines to be 1.0-1.5x the diameter of my nozzle due to Z-offset inaccuracy across the entire 450mmx450mm bed. These settings could also be used by the community to achieve their favorite flavor of raft, as with these settings, operators can tune their rafts in any that best suits their operating conditions.

**Describe alternatives you've considered**
I may be inexperienced with Super Slicer, but I as of yet have found no alternative to these issues other than manually increasing the flow rate of the first layer of the raft, or slightly clipping the wipe tower brim into the model raft.

**Additional context**
N/A

supermerill, 2024-02-15:
1) raft settings

> Raft Top Layer Height

it's the raft interface layer height, no?

> Raft Middle Layer Height

It's the raft layer height no?

> Raft Base Layer Height
> Raft Base Line Width
> Raft Base Layer Quantity

It's the first layer height.
first layer width
raft first layer quantity

> Raft Base Line Spacing
? it's computed from the width & quantity, no?

Then, for the middle raft, it's the support settings, and for the top raft, it's the support interface settings.

Is it really needed to duplicate those?

2) raft for wipe tower
good idea.

Erabior, 2024-03-27:
I apologize for the delay in responding to this. I believe most of the control I am looking for is, as you pointed out, already implemented. I will note however I would likely have never found the relevant settings due to their location in the UI. I feel like any setting that would affect the raft should be under the raft category. Otherwise it will have to pour through all of the settings to find the parameter you're looking for. Which could be fairly daunting for some.