supermerill/SuperSlicer #4127: error in volume, can't serialize fill_pattern: '3dhoneycomb' exist

enzoescipy, 2024-02-14:
### What happened?

I am using superslicer 2.5.59 currently. I was building my project with the two kinds of OBJ files imported. One was for the main body, and other was for the modifier.

The main body and modifier was well worked and I was about to print it, but I found that my infil pattern applied on my modifier was not saved as I wanted. so I re-set the infil for the modifier and found that my modifier could not set the infil patern setting.

Other setting adding was worked properly. only for the "infil pattern" modifier added by the 

right click on modifier object> infil > save

then it dissappear like magic!

### Project file & How to reproduce

[crash_log.txt](https://github.com/supermerill/SuperSlicer/files/14274491/crash_log.txt)
[ERROR_FILE_TO_SEND_TO_MERILL_PLZZZZ.txt](https://github.com/supermerill/SuperSlicer/files/14274492/ERROR_FILE_TO_SEND_TO_MERILL_PLZZZZ.txt)

### Version

2.5.59

### Operating system

windows 11 pro

### Printer model

not included with hardware problem

enzoescipy, 2024-02-14:
this is my crashed project.
[crashed_project.zip](https://github.com/supermerill/SuperSlicer/files/14274513/crashed_project.zip)

supermerill, 2024-02-14:
Thank you for reporting that.
It's an old issue that I can't reproduce sometimes, when you're using a pattern in a modifier, it fails to write it.
I put these logs and error message to have some inputs on it, but as it's a random rare event, I don't have much data.

A question for you:
 In your project, before saving it, did you set the sparse infill pattern in your modifiers?
 If yes, what was the pattern you used?
 
 As the safeguard detected a fail, it removed it from your 3mf, so I can't know. The log says you had set a 3dhoneycomb on each volume, but I want to be sure it's exactly that.

supermerill, 2024-02-15:
I added some more logs into the current release. It will be appreciated if you can use this new one and send me again the log when they appear again.
I think maybe I can see how it's failing to write, but still not the root cause.

enzoescipy, 2024-02-16:
I changed the version to 2.5.59.8, the latest release.
You know what? the issue solved. for the uploaded files of course, and my original project files the 3D honeycomb pattern just normally saved without causing crashes.

Dose it intended by developers?

enzoescipy, 2024-02-16:
[crashed_project_cured.zip](https://github.com/supermerill/SuperSlicer/files/14304004/crashed_project_cured.zip)

For sure, this is the cured project file. Without any error log file created, .3mf file just normally saved with 3D honeycomb modifier.

supermerill, 2024-02-20:
> Dose it intended by developers?

The issue seems random, or at least need a questionable chain of event to occur.

enzoescipy, 2024-02-20:
At least, I couldn't reproduce this bug from now on. good for me, but also I have to wonder with it from now on...

closed 2024-02-20 by enzoescipy