supermerill/SuperSlicer #4128: Missing recent printer profiles? 2.5.59.7

TLMcNulty, 2024-02-14:
**Bug description**
As I understand it the Prusa Mk4 should have a profile available in the wizard automatically? I can't find this profile, and exporting one from prusaslicer is not importing what I would expect. I've set the application to update profiles automatically, and tried to import/export multiple times, but selecting the imported profile wipes the material and quality settings, and gives me dubious print time estimates. Selecting filament to be added to the dropdowns doesn't populate the fields. 

**Expected behavior**
Select Mk4 in profile wizard, slice with filament and quality dropdown settings as normal. 

!!>> Project File <<!!
[Califlower v29-bug-proj.zip](https://github.com/supermerill/SuperSlicer/files/14281222/Califlower.v29-bug-proj.zip)

**To Reproduce**
Steps to reproduce the behavior:
1. Open the configuration wizard
2. Select Prusa FFF
OR
1. Import exported prusa config
2. Attempt to change quality settings

**Screenshots**
[image]
[image]
[image]

**Desktop (please complete the following information):**
 - OS:  Windows

 - Version 2.5.59.7
 - Printer model Prusa i3 Mk4

**Additional context**
This has got to be user error, but I can't find docs to help me solve, or other relevant issues.

neophyl, 2024-02-14:
Importing a Printer profile (such as importing the MK4 from PS) ONLY imports that single profile.  It wont import the associated print and filament profiles.  Unless all of those have been put into a specific Config Bundle.  A config bundle can contain several profiles of all the 3 types.

As to if the MK4 should be in the config wizard at this point, that's a question for Merill.  Of course once it is added to the wizard properly then all the associated print and filament profiles should also be added with it.

TLMcNulty, 2024-02-14:
Thanks, filament and quality do come in with an exported config bundle, but the printer preset gets set to my first user preset and doesn't seem to pull in the printer name. 

Also this message comes up but I don't think that's much of a problem 
[image]

supermerill, 2024-02-14:
It's not much of a problem.
For the mk4, as prusa doesn't have a profile for the 2.5 but for newer version, I was thinking to adding it with the 2.7 merge.

If you want, you can try to add it yourself:
 . replace the `prusa.ini` & `prusa.idx` & the `prusa` directory in `resources/profiles` by the one that comes with prusaslicer
 . edit the `translate_from_prusa.py` to put `prusa.ini` as the `filename_in` and `filename_out`
 . run the script with `python translate_from_prusa` (you need python installed)
 . then open SuperSlicer, it should ask you if you want to upgrade
But the script only goes over the big stuff. It may still not work, or at least optimally.

supermerill, 2024-02-14:
Ah yes, it can duplicate some keys. I forgot.