supermerill/SuperSlicer #4130: strange layer after printed support, 0.2 (1layer) gap

wildwick69, 2024-02-14:
### What happened?

after a printed support with 1 layer distance to continue with the overhang the layer surface is misbuilt and doesn't make sense compared to Prusaslicer

### Project file & How to reproduce

[Overhang_Layer_Error.zip](https://github.com/supermerill/SuperSlicer/files/14289210/Overhang_Layer_Error.zip)

### Version

2.5.59.x

### Operating system

w10

### Printer model

Voron Trident

supermerill, 2024-02-15:
1) A bug : `No perimeters on bridge area` is shown as deactivated with arachne, but it's not true.
You should get back to classic perimeter and deactivate it.
If this algorithm fill only that, it's because it found that going more on the right side, there is nothing to support the bridge.
(still, it goes also too far on the left side). imo, it can be improved (and the bridging algo also) to create a second fill, with a different orientation.
I have alredy this alogrithm for `dense infill`. Maybe I can reuse it for bridging.... hye, it's a good idea, thanks.

2) You don't have overhangs activated, and prusa is, intended?

here your project with two-three settings changed:
[4130_Top_Idler_Left_BZV31_v2-2_fixed.zip](https://github.com/supermerill/SuperSlicer/files/14293874/4130_Top_Idler_Left_BZV31_v2-2_fixed.zip)

wildwick69, 2024-02-15:
many thanks :-)