supermerill/SuperSlicer #4131: With zero top bottom layer, First layer's sparse infill is not aliend with the rest of the print

Gray-Stone, 2024-02-15:
### What happened?

I have a model with two objects, the middle object is used as a modifier which set bottom and top solid layers to zero, so I can get the exposed infill.

On the generated infill, the first layer's infill is on a different scale compare to the rest (happens on all types of infill shape and density).  

[image]

### Project file & How to reproduce

[bug-report-combined-model.zip](https://github.com/supermerill/SuperSlicer/files/14293277/bug-report-combined-model.zip)

The model file is also available on printables. The origional 3mf is for prusa slicer, and when opening with super slicer, the filament setting, printer setting seems all get dragged in as well.

### Version

2.5.59.7

### Operating system

ubuntu 22.04

### Printer model

Voron 2.4 AB.

supermerill, 2024-02-15:
That is because the width of the extrusion isn't the same for the first layer.
So it doesn't compute the exact same shape, to have the same coverage.
if you set your fist layer width the same value as your infill width, it's aligned.

Unsure of what to do.

Gray-Stone, 2024-02-15:
Thank you for the speedy response. 

Wonder if it's possible to force the alignment on sparse infill feature regardless of line-width (with a toggle option). Or would it be too much of a specific feature?  

Feel free to close the issue if you think this should be "fixed" through line width setting.

supermerill, 2024-02-15:
I guess I can make a more generic feature like "align sparse infill" that works for all infill, all layers.
A bit more work to also apply it for other infill, but I can see a use.

supermerill, 2024-02-15:
you can answer here in some months if it's still not implemented.

GroovybandLive, 2024-02-17:
In my opinion the best way to handle this issue (that I too recently stumbled upon) is to add a settable flag (perhaps under the "Width and Flow" page) that says "Always use infill width even on first layer". When this flag is ON then the (sparse) infill width takes precedence over the first layer width, just for the infill, when the infill is printed on first layer because the bottom layers are zero.
In this way you can have on first layer perimeters and solid fill regions and infill regions (through modifier solids) each one with different line widths.
To make things more orthogonal you could also add a similar flag for "perimeters", "external perimeters" and "solid fills" widths.
With 4 flags you can fully customise the behavior of your widths (internal/external perimeter, solid fill, sparse infill) even on first layer. The first layer width is a catch all setting for when no customisation is done.

supermerill, 2024-02-28:
I'll add the setting now, as most people tags this issue as a bug.

closed 2024-05-21 by supermerill