supermerill/SuperSlicer #4135: SuperSlicer crashes when generati

boconnell89, 2024-02-16:
### What happened?

Upon trying to slice anything in the newer versions of SuperSlicer, I get the following error:
```
Failed to rename the output G-code file from C:\Users\foo\AppData\Local\Temp\.225364.gcode.tmp.postprocess to C:\Users\foo\AppData\Local\Temp\.225364.gcode.tmp
Is C:\Users\foo\AppData\Local\Temp\.225364.gcode.tmp.postprocess locked? (gcp)permission denied
```
Note, I do not have any post-process scripts.

### Project file & How to reproduce

This has the same issue with all my (working in previous versions) FDM printers.  The default SLA printer doesn't have the same issue.

### Version

2.5.59.8

### Operating system

windows 10

### Printer model

Sovol SV07

legend069, 2024-02-17:
is your c:/users/``foo`` path correct?
``foo`` is generally used as a placeholder for a variable name, but if your home dir is correct then it's name ``foo`` shouldn't be a problem.

also please attach your project file.

supermerill, 2024-02-22:
Sometimes, the os don't let the program write files. Can't do much about it. Prusa already harden the renaming routine a lot to prevent any failure when possible, with hundreds of retries with different names.

I'll add some debug information for the next nightly.
If you start it like that in a terminal: `superslicer_console.exe -loglevel=5` (win+R -> type 'cmd' -> Enter -> darg & drop superslicer_console.exe into the terminal window -> type ' -loglevel=5' in the termnal -> Enter), then you will be able to see exactly what's wrong with the os and the files.