supermerill/SuperSlicer #4136: Inserting G1 before start macro, causing error on printer

jmceara, 2024-02-16:
**Bug description**
Superslicer is putting some GCode before "Start G-Code", causing error while printing (Axis is not homed). More exactly this code:
```G1 Z0.25 F27000
M107```

**Expected behavior**
Don't inset anything else (movements gcode, at least) before "Start G-Code"

[ValentinesHeart.zip](https://github.com/supermerill/SuperSlicer/files/14311976/ValentinesHeart.zip)

**To Reproduce**
Open project, slice and export gcode.

**Screenshots**
![image](https://github.com/supermerill/SuperSlicer/assets/3300517/fec50362-58e7-450f-8f9b-424a4ad11fa1)

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version 2.5.59.8
 - Voron v2.4 R2 with Klipper

legend069, 2024-02-17:
disabled wipe_tower and the extra z movement command is gone.
re sliced with 2.5.59.7 bug is still present.

i can't replicate it with my config using a single extruder.(only the ``m107`` command is inserted before the start_gcode)

jalanjarosz, 2024-02-25:
I can replicate this behavior in v2.5.59.8 Windows when Wipe Tower is active.

Current fix is to use G-Code substitution to remove extra G1 Z move, but that also removes a needed one.
[image]

Inserted at the end of Start G-Code the following using a slightly different feed rate:
```
; Drop down to first_layer_height due to removing extra Wipe Tower Z-Move
; by using G-code substituation which remove two G1 Z0.2 F720 moves
; one inserted by the Wipe Tower, and one that is needed.
G1 Z[first_layer_height] F600
```

supermerill, 2024-02-26:
I hcnage dthe buffeeing for the start gcode. I missed the part with the sequential printing with wipe tower

closed 2024-05-21 by supermerill