supermerill/SuperSlicer #4138: Overlap for gap fill is still affected by solid_infill_overlap

raspijam, 2024-02-17:
### What happened?

### Expected behavior:
slicing a 1.5mm thick object with 0.5mm wide perimeters AND gap_fill_overlap set to 0 would result in the gap fill extrusion being 0.5mm wide

[image]

### Actual behavior:
the resulting gap fill is 0.543mm wide

[image]

### Possible fix
setting solid_infill_overlap to 0 makes the gap fill extrusion 0.5mm wide, which shows that gap fill overlap is still affected by it

### Project file & How to reproduce

included is a project with the default superslicer profile, extrusion width set to 0.5mm and gap_fill_overlap set to 0
[gap_fill_overlap.zip](https://github.com/supermerill/SuperSlicer/files/14319680/gap_fill_overlap.zip)

### Version

2.5.59.8

### Operating system

Windows 10

### Printer model

all

raspijam, 2024-02-17:
### Additional context: 

- with gap_fill_overlap and solid_infill_overlap set to 0 the gap fill extrusion is .586mm wide, which would be consistent with Slic3r perimeter overlap rules

- with gap_fill_overlap set to 100% and solid_infill_overlap set to 0 the gap fill extrusion is .543mm wide, indicating that **the amount of how much the gap fill overlaps is a sum of gap_fill_overlap and solid_infill_overlap instead of just gap_fill_overlap**

supermerill, 2024-02-28:
nice catch.
was using gap fill spacing instead of the perimeter spacing to create the gap fill area.
and then i was using the solid infill overlap instead of the gap fill overlap to fill the area
Fixed for next nightly.

closed 2024-05-21 by supermerill