supermerill/SuperSlicer #4140: failed to start arcwelder script (sum script in temp file that doesnt exist) error 2

skullcrushercmd, 2024-02-18:
### What happened?

lastest version of superslicer, was doing fine but changed some simple settings in the profiling and this came up when I wanted to export

Failed starting the script ArcWelder.exe C:\Users\tooni\AppData\Local\Temp\.SuperSlicer.upload.df81-7e42-5eb2-3e60, Win32 error: 2

idk how to fix but prusaslicer works fine

### Project file & How to reproduce

unsure as it does it to any stl I slice and try to export

### Version

2.5.59

### Operating system

windows 11

### Printer model

anycubic kobra 2

jbeima, 2024-02-23:
May I ask a possibly strange question. Since 2.5.59 now can generate G2/G3 arc moves natively, is there a reason to still use ArcWelder? Just wondering? Is there something ArcWelder does that the now built in generation does not do?

skullcrushercmd, 2024-02-23:
> May I ask a possibly strange question. Since 2.5.59 now can generate G2/G3 arc moves natively, is there a reason to still use ArcWelder? Just wondering? Is there something ArcWelder does that the now built in generation does not do?

I'm kina new to 3d printing so I don't really know. Idk what arcwelder was even used for 💀

supermerill, 2024-02-24:
How can i reproduce the issue?

skullcrushercmd, 2024-02-24:
> How can i reproduce the issue?
> 

I don't really know because I forgot, tried reinstalling but didn't work. I'd be happy to help with debugging